MarginallyClever/Makelangelo-software #478: Popup Error on retransmission

coliss86, 2022-01-26:
**Steps to make the bug happen**
I'm trying a Bluetooth adapter HC06 connected to the robot and sometimes, the following popup appears and blocks the robot.
I think this is caused by error in transmissions.
[image]

If I click OK, the popup appears multiple times and finally the print continues.

**What was supposed to happen**
I think the popup should not appears when the resend works.

**Platform (please complete the following information):**
 - Version: master
 - OS: os x
 - Robot model (if any): diy with *Bluetooth* adapter

**Log file**
Some more logs: 
```
2022-01-26 12:32:56,219 TRACE c.m.m.p.p.MarlinInterface - sending 'N314 G1 X50.921 Y89.088 F3000.0*90' 
2022-01-26 12:32:56,258 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 12:32:56,258 TRACE c.m.m.p.p.MarlinInterface - sending 'N315 G1 X51.386 Y88.965 F3000.0*86' 
2022-01-26 12:32:56,267 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 12:32:56,267 TRACE c.m.m.p.p.MarlinInterface - sending 'N316 G1 X52.261 Y88.524 F3000.0*87' 
2022-01-26 12:32:56,320 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 12:32:56,322 TRACE c.m.m.p.p.MarlinInterface - sending 'N317 G1 X51.634 Y88.416 F3000.0*81' 
2022-01-26 12:32:57,244 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 12:32:57,245 TRACE c.m.m.p.p.MarlinInterface - sending 'N318 G1 X51.016 Y88.250 F3000.0*92' 
2022-01-26 12:32:57,246 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 12:32:57,246 TRACE c.m.m.p.p.MarlinInterface - sending 'N319 G1 X50.006 Y87.833 F3000.0*93' 
2022-01-26 12:32:58,168 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 12:32:58,168 TRACE c.m.m.p.p.MarlinInterface - sending 'N320 G1 X48.856 Y87.204 F3000.0*93' 
2022-01-26 12:32:58,169 TRACE c.m.m.p.p.MarlinInterface - received 'Error:checksum mismatch, Last Line: 315' 
2022-01-26 12:32:58,169 ERROR c.m.m.p.p.MarlinInterface - Error from printer 'Error:checksum mismatch, Last Line: 315' 
```
**Additional context**
Add any other context about the problem here.
If applicable, add screenshots to help explain your problem.

coliss86, 2022-01-26:
I think that an error of the kind `Error:checksum mismatch` should not trigger a popup.

@i-make-robots WDYT ?

Keepthefunk, 2022-01-26:
here is my log - hope it helps!
[makelangelo.zip](https://github.com/MarginallyClever/Makelangelo-software/files/7942657/makelangelo.zip)

coliss86, 2022-01-26:
More details: 
on first error, Makelangelo resends the wrong line, it should resend 439 but logs says it resends 440
```
2022-01-26 16:42:14,286 TRACE c.m.m.p.p.MarlinInterface - sending 'N436 G1 X-19.910 Y23.662 F3000.0*125' 
2022-01-26 16:42:14,286 TRACE c.m.m.p.p.MarlinInterface - sending 'N437 G1 X-20.134 Y23.899 F3000.0*114' 
2022-01-26 16:42:14,286 TRACE c.m.m.p.p.MarlinInterface - sending 'N438 G1 X-20.353 Y24.186 F3000.0*126' 
2022-01-26 16:42:14,292 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 16:42:14,292 TRACE c.m.m.p.p.MarlinInterface - sending 'N439 G1 X-20.565 Y24.518 F3000.0*127' 
2022-01-26 16:42:14,293 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 16:42:14,293 TRACE c.m.m.p.p.MarlinInterface - sending 'N440 G1 X-20.768 Y24.892 F3000.0*113' 
2022-01-26 16:42:14,332 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 16:42:14,332 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 16:42:14,333 TRACE c.m.m.p.p.MarlinInterface - sending 'N441 G1 X-21.051 Y25.517 F3000.0*125' 
2022-01-26 16:42:14,334 TRACE c.m.m.p.p.MarlinInterface - sending 'N442 G1 X-21.303 Y26.207 F3000.0*127' 
2022-01-26 16:42:14,344 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 16:42:14,344 TRACE c.m.m.p.p.MarlinInterface - sending 'N443 G1 X-21.518 Y26.946 F3000.0*124' 
2022-01-26 16:42:14,350 TRACE c.m.m.p.p.MarlinInterface - received 'Error:checksum mismatch, Last Line: 439' 
2022-01-26 16:42:14,350 ERROR c.m.m.p.p.MarlinInterface - Error from printer 'Error:checksum mismatch, Last Line: 439' 
2022-01-26 16:42:14,350 TRACE c.m.m.p.p.MarlinInterface - received 'Resend: 440' 
2022-01-26 16:42:14,350 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 16:42:14,350 TRACE c.m.m.p.p.MarlinInterface - sending 'N440 G1 X-20.768 Y24.892 F3000.0*113' 
2022-01-26 16:42:14,360 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 16:42:14,360 TRACE c.m.m.p.p.MarlinInterface - sending 'N441 G1 X-21.051 Y25.517 F3000.0*125' 
2022-01-26 16:42:14,371 TRACE c.m.m.p.p.MarlinInterface - received 'Error:Line Number is not Last Line Number+1, Last Line: 439' 
2022-01-26 16:42:14,371 ERROR c.m.m.p.p.MarlinInterface - Error from printer 'Error:Line Number is not Last Line Number+1, Last Line: 439' 
2022-01-26 16:42:28,310 TRACE c.m.m.p.p.MarlinInterface - received 'Resend: 440' 
2022-01-26 16:42:28,310 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 16:42:28,310 TRACE c.m.m.p.p.MarlinInterface - sending 'N440 G1 X-20.768 Y24.892 F3000.0*113' 
2022-01-26 16:42:28,311 TRACE c.m.m.p.p.MarlinInterface - received 'Error:Line Number is not Last Line Number+1, Last Line: 439' 
2022-01-26 16:42:28,311 ERROR c.m.m.p.p.MarlinInterface - Error from printer 'Error:Line Number is not Last Line Number+1, Last Line: 439' 
2022-01-26 16:42:29,069 TRACE c.m.m.p.p.MarlinInterface - received 'Resend: 440' 
2022-01-26 16:42:29,069 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 16:42:29,069 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 16:42:29,069 TRACE c.m.m.p.p.MarlinInterface - sending 'N440 G1 X-20.768 Y24.892 F3000.0*113' 
2022-01-26 16:42:29,069 TRACE c.m.m.p.p.MarlinInterface - sending 'N441 G1 X-21.051 Y25.517 F3000.0*125' 
2022-01-26 16:42:29,070 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 16:42:29,070 TRACE c.m.m.p.p.MarlinInterface - sending 'N442 G1 X-21.303 Y26.207 F3000.0*127' 
2022-01-26 16:42:29,071 TRACE c.m.m.p.p.MarlinInterface - sending 'N443 G1 X-21.518 Y26.946 F3000.0*124' 
2022-01-26 16:42:29,114 TRACE c.m.m.p.p.MarlinInterface - received 'Error:Line Number is not Last Line Number+1, Last Line: 440' 
2022-01-26 16:42:29,114 ERROR c.m.m.p.p.MarlinInterface - Error from printer 'Error:Line Number is not Last Line Number+1, Last Line: 440' 
2022-01-26 16:42:29,721 TRACE c.m.m.p.p.MarlinInterface - received 'Resend: 441' 
2022-01-26 16:42:29,721 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 16:42:29,721 TRACE c.m.m.p.p.MarlinInterface - received 'echo:Unknown command: "*125"' 
2022-01-26 16:42:29,721 TRACE c.m.m.p.p.MarlinInterface - sending 'N441 G1 X-21.051 Y25.517 F3000.0*125' 
2022-01-26 16:42:29,721 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-01-26 16:42:29,722 TRACE c.m.m.p.p.MarlinInterface - sending 'N442 G1 X-21.303 Y26.207 F3000.0*127' 

```

coliss86, 2022-01-26:
Just tried with a usb cable, same error so not related to HC06 Bluetooth

i-make-robots, 2022-01-26:
`MarlinInterface` is perfectly able to handle checksum mismatches so they are not fatal errors.  It is also (nearly always) able to handle bad line number (seen in #479) without fatal error.
When did these become a popup?  whatever we "fixed" by adding a popup should be re-examined.

coliss86, 2022-01-26:
yes, a mismatches and a resend should not be trigger a popup. I'm testing an old version of Makelangelo to find when this mismatch error starts to happen.

i-make-robots, 2022-01-26:
mismatch errors happen all the time.  serial communications are unreliable.  the mismatches are not new.  the popup is new.

PPAC37, 2022-01-26:
> Makelangelo resends the wrong line, it should resend 439 but logs says it resends 440

the marlin tell us 
"Error: ..."
then
"Resend: ...."
As the error is not fatal (!)
This is the number line in the "Resend: ..." we have to resend ...

> 2022-01-26 16:42:14,350 TRACE c.m.m.p.p.MarlinInterface - received 'Error:checksum mismatch, Last Line: 439'
> 2022-01-26 16:42:14,350 ERROR c.m.m.p.p.MarlinInterface - Error from printer [...]

But as the "marlin buffer" contains "pre buffered line" (already send ) to get in track multiple "Resend: ..." will be send ( for each line buffered in the marlin ... ) until it "stabilize" ...

Le "Error:checksum mismatch" est courant dans la communication série USB (parasitage de fil ... il afut préférer un cable USB blindé et court voir éviter les grosse soudure qui capte le onde élèctromagnétique dans le montage)

Also "ok" Marlin have better than the simple 'ok' see https://marlinfw.org/docs/configuration/configuration.html#advanced-ok
and for more https://github.com/MarginallyClever/Marlin-polargraph/blob/bugfix-2.0.x/docs/Queue.md#synchronization

... so no error dialog to show as you say. ( if i'm not missing something else ...)

But while testing, I found another problem (hopping only on my linux ubuntu ): if I connect, then disconnect under Makelangelo (you must have already opened a connection once and closed it without leaving Makelangelo), then if I connect under Repetier Host (but don't disconnect), Makelangelo can connect... and it mixes up communications (like java lib issue for serial port, mishandled semaphore story...)
[image]

Also but not serious, just sad for me, Makelangelo does not know how to use a g-code file with M300s (to make "music" when you have activated the SPEAKER in marlin configuration.h )
[mario.gcode.txt](https://github.com/MarginallyClever/Makelangelo-software/files/7945872/mario.gcode.txt) playable under http://ddrboxman.github.io/RepRapRingtone/

coliss86, 2022-01-26:
ok I'll fix it asap

> But while testing, I found another problem (hopping only on my linux ubuntu ): if I connect, then disconnect under Makelangelo (you must have already opened a connection once and closed it without [...]

Same on a mac, it is due to the serial lib https://github.com/java-native/jssc which doesn't not opened exclusively the port. Even with the flag `JSSC_NO_TIOCEXCL` it does not open the port exclusively, cf https://github.com/java-native/jssc/blob/master/src/main/java/jssc/SerialPort.java#L208

PPAC37, 2022-01-26:
> Same on a mac, it is due to the serial lib https://github.com/java-native/jssc which doesn't not opened exclusively the port. Even with the flag `JSSC_NO_TIOCEXCL` it does not open the port exclusively, cf https://github.com/java-native/jssc/blob/master/src/main/java/jssc/SerialPort.java#L208

Thank you, it's not just my OS ! I'm reassured. ! :) 
I cross my fingers that in a next version of "jssc" the problem will be corrected... (my knowledge of "low level" operating systems is very far away...)

i-make-robots, 2022-01-26:
@PPAC37 The "last line" means "last line received OK".  So the line to send is the first line that fails, which is `(last ok)+1`.
Makelangelo software can send any string to a serial connection.  If you want to write a mod that sends M300s, let's discuss it and figure out how.

Repetier not using exclusive is their problem.  I remember long ago trying JSSC without success.  If you get it working... cool.  necessary?  *shrug*  I think our time would be better spent building cool new conversion styles and making the UI more intuitive.

coliss86, 2022-01-27:
https://github.com/MarginallyClever/Makelangelo-software/commit/c3538644617f7a2043ab374755a58a2bd1ec37a9 fix the problem ! thx :)

closed 2022-01-27 by coliss86

Keepthefunk, 2022-01-27:
Works like a charm! Thanks!