MarginallyClever/Makelangelo-software #482: Add a new Baudrate dropdown menu

coliss86, 2022-01-27:
**Is your feature request related to a problem? Please describe.**
I'm setting up a new HC06 bluetooth adapter which speed cannot get higher than 115200.
So I need to specify it but I don't where to put this settings.
The most evident place is next to the dropdown menu of port: 
[image]

But it will break the `TCPConnection` which is not concerned by this settings.
Maybe a better place would be in the `PlotterSettingsPanel`
[image]

Where do you think is the best place ?

i-make-robots, 2022-01-28:
No!  One baud rate to rule them all!

i-make-robots, 2022-01-28:
Get a better bluetooth adapter :)

coliss86, 2022-01-28:
So you don't want this parameter ?

i-make-robots, 2022-01-28:
I don't.  I want when-apple-was-good simplicity.

coliss86, 2022-01-29:
It is a common configuration for software that communicates with serial port. It is even available in pronterface, octoprint...
We can select by default `250000` and let the user select a different speed. Why not offering this possibility ?

PPAC37, 2022-01-29:
For me also it seems more practical to have the choice of the baudrate modifiable by the user.

Unless you have to reserve it for a paid pro version or if it's to encourage people to get into java development so that they can change it themselves:) 

or possibly a command line argument when launching from a command line (if you don't want to have to add something to modify it in the graphical interface)

i-make-robots, 2022-01-31:
Fine.  I would put the baud rate on the connect dialog.  The default should stay 250000.

i-make-robots, 2022-02-06:
In the past I had connection dialog pop-ups because I wanted to be able to seamlessly switch to TCP/IP in case in future there was support for wifi in the robot.  nothing on the near side of TextInterfaceToNetworkSession (inside makelangelo.*) should know about what kind of connection is available.  nothing on the far side (inside communications.*) should know about the makelangelo app.  At least... that was the idea.

coliss86, 2022-02-06:
Ok I see, I'll first create something that works and then refactor it to meet your goals

closed 2022-02-14 by i-make-robots