MarginallyClever/Makelangelo-software #483: LoadGcode with flavor choise ( marlin-polargraphe gcode vs makelangelo-firmware gcode)

PPAC37, 2022-01-28:
**Is your feature request related to a problem? Please describe.**
To be able to loadGcode for marlin-polargraphe ( as the curent implementation dont use G0 to penUp and G1 to penDown turtle ... but the Z value ( adapted for makelangelo-firmware gcode but not for the marlin-firmware gcode ...) )

https://github.com/MarginallyClever/Makelangelo-software/blob/master/src/main/java/com/marginallyclever/makelangelo/makeArt/io/vector/LoadGCode.java

**Describe the solution you'd like**
2 interpretation 

1 only for the marlin-polargraphe flavored gcode

adding something like 
G0 -> PenUp
G1 -> PenDown
( and removing the PenUp/Down on Z change )
to get the result (on the right) and not  the result (on the left, only moves)
[image]

And
1 only for Makelangelo-firmware gcode
Keep the actual LoadGcode version 

**Describe alternatives you've considered**
No post compatibilitie. Adapt to correctly load gcode marlin-polargraph.

**Additional context**

i-make-robots, 2022-01-28:
Mmm... I think it's easier to write a single app that does the string replacement for the few people who need it.  should auto detect which type of file it is looking at and present the user with a confirmation button.  Probably doesn't need a progress bar.  Might want to have dialog for up and down angles.  But I don't think it belongs in Makelangelo Software.

PPAC37, 2022-01-28:
I hadn't thought of the external transformation program. Thank you.

In fact, the only flaw with the current implementation of Makelangelo is that it currently generates .gcode for marlin-polargraph and interprets with LaodGCode .gcode for Makelangelo-polargraph.
Seems inconsistent to me...
But it's not a priority. So I'm closing the issue here and I'm making one on my fork so as not to forget the days when I would have done everything I need to end up here.)

closed 2022-01-28 by PPAC37

i-make-robots, 2022-01-28:
> In fact, the only flaw with the current implementation of Makelangelo is that it currently generates .gcode for marlin-polargraph and interprets with LoadGCode .gcode for Makelangelo-polargraph.

so it writes and reads the same format?  I am confused why that is a bad thing.

PPAC37, 2022-01-28:
>so it writes and reads the same format? I am confused why that is a bad thing.

Sorry i have poorly explain myself.

Not currently save gcode in marlin flavor but read as polorgraph flavor (to understand you have to try saving in GCode from Makelangelo then re-read this .gcode file with Makelangelo...)
[image]
[image]

PPAC37, 2022-01-28:
Oups bad second image
the one i get
[image]

i-make-robots, 2022-01-28:
see https://github.com/MarginallyClever/Makelangelo-software/pull/487