MarginallyClever/Makelangelo-software #484: Disable the popup of the pen change color when the robot don't have a screen

coliss86, 2022-01-28:
**Is your feature request related to a problem? Please describe.**
When the robot don't have a screen, the `M0` command is not understood by it and it will continue to draw, but on Makelangelo, a popup appears telling the user to change the pen.

**Describe the solution you'd like**
Add a new feature to the PlotterControlSettings to indicate if a screen is connected and then show or not this popup

**Describe alternatives you've considered**
none

**Additional context**
none

PPAC37, 2022-01-28:
On print control applications with 3D printers, when printing from these software there is the possibility of managing user interactions on the host side (eg for M0 and M600 (filament change) and ... any command that requires a user interaction. (or without screen there is the possibility to activate in the marlin firmware EMERGENCY_PARSER to make the interaction via the host ...))

https://marlinfw.org/docs/gcode/M000-M001.html

>Without an LCD controller or EMERGENCY_PARSER this command is ignored.  
With EMERGENCY_PARSER enabled the M108 command can be used to continue.

Or with OctoPrint there is an evaluation on the host side before sending each gcode and for example an M0, the M0 is not even sent to the printer, but OctoPrint stops sending (creating the pause in the host side with this ) and waits for the user to respond via the OctoPrint interface.

i-make-robots, 2022-01-28:
alt 1: get a screen
alt 2: continue to ignore M0, but make sure program actually pauses while PC "please change color" dialog is open.

closed 2022-02-02 by i-make-robots