MarginallyClever/Makelangelo-software #488: "CMYK spirals" and "CMYK" artwork: Pen stays down when asking to change to next color cyan, magenta, black

Keepthefunk, 2022-01-28:
**Steps to make the bug happen**
1. Open any Image file and create artwork with "Spiral CMYK" or "CMYK"
2. click Tools - simplify
3. Plot Artwork, starting with yellow pen

**What was supposed to happen**
when yellow component is done, the plotter should _lift the pen_ and ask for the next color, cyan

**What happens**
When yellow component is done, plotter asks for next color but the pen stays down, bleeding ink in to the paper.
Same happens when asking for the next colors magenta and black. When the last color black is done, the pen lifts up fine. Should also lift when asking for cyan, magenta and black.

**Platform (please complete the following information):**
 - Version: [e.g. 7.31.0.0]
 - OS: [Windows]
 - Robot model (if any): Custom with RAMPS and Marlin-polargraph-bugfix-2.0.x

[image]

Keepthefunk, 2022-01-28:
The .log file is full of "Error: Line number is not last line number+1" errors. If you need it, I can add the file

i-make-robots, 2022-01-28:
I think I understand the problem already.  Thank you. :)

i-make-robots, 2022-01-29:
gcode before simplify:
```
; 2022-01-28 at 16:01:28 PST
G28
M0 Ready yellow and click
G0 X-176.111 Y-265.577
M280 P0 S25 T50
G1 X-175.305 Y-266.169 F3000.0
G1 X-174.498 Y-266.759 F3000.0
M280 P0 S90 T50
G0 X-174.363 Y-261.896
M280 P0 S25 T50
```
gcode after simplify:
```
; 2022-01-28 at 16:00:44 PST
G28
M0 Ready black and click
G0 X-176.111 Y-265.577
M280 P0 S25 T50
G1 X-175.305 Y-266.169 F3000.0
M280 P0 S90 T50
G0 X-176.111 Y-265.577
M280 P0 S25 T50
G1 X-175.305 Y-266.169 F3000.0
M0 Ready yellow and click
M280 P0 S90 T50
```
note the pen up has moved to immediately AFTER the M0.

PPAC37, 2022-01-29:
for me this is the way we tranlate TurtleMove.TOOL_CHANGE in SaveGCode ( should check if the plotter have a servo and if yes then add the pen Up and Down ...

> Also i just noticed it reorders the colors. WTF? That should not happen.

This is usfull for the Generator_GraphPaper ... or ( as in turtle it do something like : black grid horizontal, then red grid, then black grid vertical in the generator 
and doing a reorder allow to have 1 change color less in the gcode )

> note the pen up has moved to immediately AFTER the M0.

That i dont understande.

i-make-robots, 2022-01-29:
@PPAC37 Um... the graph paper system does all the red THEN all the black.  There is no black-red-black.  
```java
		turtle.setColor(new ColorRGB(255,0,0));
		lines(turtle,10,0);
		lines(turtle,10,90);
		turtle.setColor(new ColorRGB(0,0,0));
		lines(turtle,100,0);
		lines(turtle,100,90);
```
Maybe you're seeing it differently because of issue #488.

Simplify and Reorder both convert a turtle into a set of LineSegment2D.  Each one has a single drawing-only move.  The travel moves are only implied.  

https://github.com/MarginallyClever/Makelangelo-software/blob/c4b369c1713061acd0302a6fd4fdc881b40fe1d7/src/main/java/com/marginallyclever/makelangelo/turtle/Turtle.java#L386-L388 When the set is turned back into a Turtle it looks for changes in color FIRST and does the travel move SECOND.  This is different from the way CMYK generates turtle moves - lift, then color change.

> TurtleMove.TOOL_CHANGE in SaveGCode

A good start.  Would not fix case where the plotter is connected via serial.

PPAC37, 2022-01-29:
My excuses. I must be confusing things.
(Because I remembered having noticed an unnecessary color change in the graph generator when saving the rendering in .gcode but reordering it before the SaveGcode fixed the problem.
But that doesn't seem to be the case anymore...

Keepthefunk, 2022-01-29:
Thanks all. I am glad my first posted issue is understood. To my humble knowledge there should always be a 
`M280 P0 S90 T50` (pen lift) *before* `M0 Ready colorxxx and click` (it stops). 
So that the pen is up when the plotter is stopped.

PPAC37, 2022-01-29:
@Keepthefunk  The fix should be coming soon.

And thank you, it makes me think to put in the list of things to do one day maybe if not too complicated, the fact of having a way to take into account the pen that diffuses the ink on certain paper (and therefore the speed of displacement and pauses to be taken into account) to create style effects thanks to its ink diffusion...

Keepthefunk, 2022-01-29:
I just tested the last nightly build with CMYK spirals. Now I can use "reorder" and it does not mix up the colors, so already an improvement.
But still the pen stays down while waiting for the next color, letting ink bleed in to the paper. Tried with Simplify+reorder and simplify only.

@PPAC37 style effects with ink diffusion is a great idea. In my case I would be happy to get rid of diffusion. If only the pen would lift when waiting. At the moment I solve this problem by standing next to the plotter, wait til finished and lift the pen by hand just in time.

Keepthefunk, 2022-01-29:
Here is a simple CMYK image for testing. The file does not have too much color so the test is quick.
[image]

PPAC37, 2022-01-29:
So perfect for the pen Up done just before a pause to ask color change.

But will testing, i found that maybe 
we should use a   jSpinner1.setModel(new javax.swing.SpinnerNumberModel(1, 1, null, 1)); //to assert a positive value >= 1 in the panel

https://github.com/MarginallyClever/Makelangelo-software/blob/782f02f9dcbb4866f23274a5a21da6b34f81138c/src/main/java/com/marginallyclever/makelangelo/makeArt/imageConverter/Converter_CMYK_Panel.java#L12

and correcting  
https://github.com/MarginallyClever/Makelangelo-software/blob/782f02f9dcbb4866f23274a5a21da6b34f81138c/src/main/java/com/marginallyclever/makelangelo/makeArt/imageConverter/Converter_CMYK.java#L37-L40

to somethingue like that to prevent any trouble ... ( like if you put a "0" as passes value in the panel ... => kind of freeze the application ...)
```
public void setPasses(int value) {
		if(value<1) 
		    passes=1;
		else 
		    passes=value;
	}
```

Keepthefunk, 2022-01-30:
This works now! Pen lifts up before stopping and asking for next pen. THANKS!

closed 2022-01-30 by Keepthefunk