MarginallyClever/Makelangelo-software #489: Plotter Controls > Advanced Controls render outside parent bounding limits

i-make-robots, 2022-01-29:
**Steps to make the bug happen**
1. Load any image or vector
2. Open plotter controls
3. expand advanced controls
4. See error

**What was supposed to happen**
tabs inside advanced control are now hanging off the edge of the window and cannot be seen.  "DRAW_LINE" in program window appears as "RAW_LINE".

**Platform (please complete the following information):**
 - Version: 7.31.x
 - Windows

**Additional context**
Also I see the plotter controls are called "robot controls" and so is the jog panel!  Two things with different jobs and the same name?!

PPAC37, 2022-01-29:
Same for me (on linux)
If we resize the width, larger then smaller then there is a placement defect 
[image]

(but if we click again on the tabs (change tab and come back) this is corrected...)

> Also I see the plotter controls are called "robot controls" and so is the jog panel! Two things with different jobs and the same name?!

sorry dont get it ... you are talking about the title of the dialogue and the title of a tab in Advanced controls  which are "identical" on my capture or something else.

i-make-robots, 2022-01-29:
> sorry dont get it ... you are talking about the title of the dialogue and the title of a tab in Advanced controls which are "identical" on my capture or something else.

Yes, exactly.  They are different, they should not be named the same.

PPAC37, 2022-01-29:
```
<string>
		<key>PlotterControls.Title</key>
		<value>**Robot controls**</value>
		<hint>title</hint>
	</string>
...
	<string>
		<key>PlotterControls.JogTab</key>
		<value>**Robot controls**</value>
		<hint>panel tab title</hint>
	</string>
```

coliss86, 2022-01-29:
I'll fix it

commit 7795c7c1 (2022-02-01): 🐛 Fix #489: set the inner component's size with bounds (#491)

* 🐛 Fix #489: set the inner component's size with bounds
* Fix NPE on windows
* Fixed !
closed 2022-02-01 by i-make-robots