MarginallyClever/Makelangelo-software #493: Files opening

coliss86, 2022-01-29:
**Is your feature request related to a problem? Please describe.**
When opening a file, you you have to click twice to have the choose file dialog : In the menu and on the popup that just appeared.
In gimp or inkscape, when you click on the "Open..." you can directly select the file.

Moreover, in gimp, by default the file filter is set to every type of file supported. here, you have to select the file type in svg for example and then you can select the file.

Besides, in the case of svg files, the LoadFilePanel is empty, so I think it is not needed.

**Describe the solution you'd like**
I would like that the LoadFilePanel don't show before the selection file window.
I also would like that the selection filter match all type of file, eg: dxf, svg, png jpeg... and be selected by default

**Describe alternatives you've considered**
none

coliss86, 2022-01-29:
Here is a first prototype of the "all file types": 
[image]

PPAC37, 2022-01-30:
i like it ! 

[image]
[image]

PR #497 (2022-02-02): 💄 Open a file chooser directly when clicking on "Open file..."
Fix #493
closed 2022-02-02 by i-make-robots