MarginallyClever/Makelangelo-software #505: Application in german when launching for the first time

coliss86, 2022-01-31:
**Steps to make the bug happen**
1. Start the app on a brand new computer: it is in german. Quite hard to find the settings to change the language ;)

**What was supposed to happen**
The default language should be english

**Platform (please complete the following information):**
 - Version: master
 - OS: windows
 - Robot model (if any): [e.g. Makelangelo 5]

**Log file**
none

PPAC37, 2022-01-31:
So no chose language dialog at the first run ?
[image]
( if i delete my ~/.java/.userPrefs/DrawBot/Language i get this . But strangly this dialgue is not seen as a "windows aplication" in my desktop manager so i dont have a spécific icone in my systeme application bar ... )

coliss86, 2022-01-31:
Nop, nothing :'(
I tried on Windows.

PPAC37, 2022-01-31:
And no log ?

https://github.com/MarginallyClever/Makelangelo-software/blob/98686f479880423c293cb2ade333754f8d3b1fcb/src/main/java/com/marginallyclever/makelangelo/Makelangelo.java#L1008-L1010

https://github.com/MarginallyClever/Makelangelo-software/blob/98686f479880423c293cb2ade333754f8d3b1fcb/src/main/java/com/marginallyclever/makelangelo/Translator.java#L59-L81

https://github.com/MarginallyClever/Makelangelo-software/blob/98686f479880423c293cb2ade333754f8d3b1fcb/src/main/java/com/marginallyclever/makelangelo/makelangeloSettingsPanel/LanguagePreferences.java#L42-L55

PPAC37, 2022-01-31:
Also not the problem here but was wondering where the prefrences are store in other OS type and version  ?

coliss86, 2022-01-31:
I'll try again asap.
thank for pointing where to look for :)

coliss86, 2022-02-11:
I did not get time to test it again

coliss86, 2022-02-17:
I tried again and was not able to reproduce the issue. Closing it.

closed 2022-02-17 by coliss86