MarginallyClever/Makelangelo-software #508: Label truncated

coliss86, 2022-02-02:
**Steps to make the bug happen**
1. Load a big image
2. Apply A CMYK transformation
3. The label on the lower right of the slider is truncated
[image]

**What was supposed to happen**
It should not be truncated

**Platform (please complete the following information):**
 - Version: master
 - OS: macos

**Log file**
none

i-make-robots, 2022-02-02:
I set fixed values for the number and had no idea how many would be there.  If I let the system dynamically choose the best size a new problem appears: as I drag the slider the field size changes, which affects the slider length, which moves the slider head on the slider.  so it can happen that you're trying to drag to a very specific number, the slider jumps, and you CAN'T hit the number you want.

i-make-robots, 2022-02-05:
I think the best solution is to allow the fields to grow but not to shrink.  so they might look funny if there's a small number in a large space... but it will work with minimum user friction.

i-make-robots, 2022-02-07:
Minimum size is ignored by something in the system.  Maybe the BorderLayout?  In any case it doesn't work.  Preferred size and maximum size don't work either, number truncated.

coliss86, 2022-02-07:
The latest solution is to pad on the right the number with space. So every time, the label is at the correct size

i-make-robots, 2022-02-08:
I doubt spaces are the same with as digits in all fonts.

coliss86, 2022-02-08:
Defining a fixed size font will ensure it. Something like : 
```
.setFont(new Font(Font.MONOSPACED, Font.PLAIN, 12));
```

coliss86, 2022-02-09:
The problem is still present. The number is `351228`
[image]

By setting the font and increasing a little the space, it solves the problem : 
[image]

```
labelRangeMin.setBorder(BorderFactory.createEmptyBorder(0, 5, 0, 5));
labelRangeMin.setFont(new Font(Font.MONOSPACED, Font.PLAIN, 12));
labelRangeMax.setBorder(BorderFactory.createEmptyBorder(0, 5, 0, 5));
labelRangeMax.setFont(new Font(Font.MONOSPACED, Font.PLAIN, 12));
Dimension d = labelRangeMin.getPreferredSize();
d.width=60;
```

i-make-robots, 2022-02-09:
Damnit, I thought I made enough room for 7 digits.

i-make-robots, 2022-02-11:
@coliss86 added your fix in 75a47cc1.  looks good!

closed 2022-02-11 by i-make-robots