MarginallyClever/Makelangelo-software #513: Some elements not translated in the current GUI

PPAC37, 2022-02-04:
Some Panel title
* [ ] PaperSettings
* [ ] PlotterSettingsPanel
* [ ] Elements of Menu "View", "Render mode"
https://github.com/MarginallyClever/Makelangelo-software/blob/c4b369c1713061acd0302a6fd4fdc881b40fe1d7/src/main/java/com/marginallyclever/makelangelo/turtle/turtleRenderer/TurtleRenderFactory.java#L14-L17

Some descriptions of some Generat Arts
https://github.com/MarginallyClever/Makelangelo-software/search?q=Learn+more&type=code

Some Buttons
* [ ] Prefrence : "OK" / "Cancel"

PPAC37, 2022-02-06:
#517 normally solves the first three points/tasks.

Will not be done 
*  * [ ] Prefrence : "OK"
*  * [ ] Prefrence : "Cancel"

because request to decompose showConfirmDialog to modify the names of the options (and it seems to me that some JREs do it on their own) https://github.com/MarginallyClever/Makelangelo-software/blob/98686f479880423c293cb2ade333754f8d3b1fcb/src/main/java/com/marginallyclever/makelangelo/makelangeloSettingsPanel/MakelangeloSettingPanel.java#L84

* And "Home" will stay like that.

---
Stay to be done 

* The "Learn More" will be falsly solve be adding a translation to "Learn More" ( but not for each text where it appears because it requires for my tastes to create too many translation keys ... and as Wikipedia has links to other languages ...)

i-make-robots, 2022-02-08:
imho the home button can be removed from cartesian buttons.  afaik nobody uses it and the home button already exists in *at least* one other place.

commit 3a494d86 (2022-02-10): fix #513 (#530)

* do nothing when center clicked.
* translate buttons and use new naming convention.
* add translation key/value