MarginallyClever/Makelangelo-software #520: Wait a little after servo move

coliss86, 2022-02-05:
**Is your feature request related to a problem? Please describe.**
A user on discord asked me to test adding a wait after the servo moved down to reduce wobbling.
I created for him a branch on my repo with a `G4 P800` and it seems to work very good for him.

So I'm proposing to add this as a new config in the plotter settings. It is deactivated by default and the user can specified an amount of time to wait after the servo moved down.

What do you think ?

**Describe the solution you'd like**
The gcode part is implemented in this PR https://github.com/coliss86/Makelangelo-software/pull/11/files

**Describe alternatives you've considered**
None

**Additional context**
None

PPAC37, 2022-02-06:
I love it, it somehow allowed to increase the firmware `#define SERVO_DELAY 300` cf : https://marlinfw.org/docs/configuration/configuration.html#servos

coliss86, 2022-02-06:
good point ! I forgot this setting.

i-make-robots, 2022-02-07:
the T value for servo speed can be adjusted in the plotter settings window.  no need to recompile anything.

PPAC37, 2022-02-07:
> the T value for servo speed can be adjusted in the plotter settings window. no need to recompile anything.

The modifications of the marlin source code specific to the marlin-polargaphe version, for me it is absolutely necessary to document them (wiki? or in the root README.md) because I do not have the reflex to look at the commits made on a branch. ..

https://github.com/MarginallyClever/Marlin-polargraph/commit/5ffe5702c7f2f1b68fc3a79ff46db822a1925e2f

carl1961, 2022-02-07:
@i-make-robots  in plotter firmware  7.31.0 I see no settings option

PPAC37, 2022-02-07:
@carl1961  as it was added not long ago, I think it is not yet in one of Makelangelo-software's "Releases", you have to either work with a version that you compile from the master branch or test a Nightly build https://github.com/MarginallyClever/Makelangelo-software/releases/tag/Nightly
You should find in the menu bar, Settings, Plotter Settings ... a "Lift delay ..."

[image]

carl1961, 2022-02-07:
@PPAC37  Thanks, the latest nightly worked. [Makelangelo-nightly-20220207-9bb29a.zip](https://github.com/MarginallyClever/Makelangelo-software/releases/download/Nightly/Makelangelo-nightly-20220207-9bb29a.zip)

[image]

carl1961, 2022-02-07:
@PPAC37 , thanks for all your help, is there any writing explaining the New settings? stepper wiring length, belt timing length etc. a example for Maklangelo 5 setting would help me a lot.

PPAC37, 2022-02-07:
@carl1961  Not easy to answer, I have not worked on the parameters of the plotter. But it seems to me that currently the size and length elements are used for display (the limited frames / border in the main view). (because they are in "hard" in the firmware ...) and have no impact on the commands sent to the machine or written in the .gcode file.
But on the other hand the speeds, acceleration, pen angle and delays seem to be taken into account to control the plotter.
 But I leave it to @i-make-robots  to confirm or not.

i-make-robots, 2022-02-07:
In the old days I used makelangelo-firmware on the robots.  MF would allow you to change the machine size with a gcode command.  so the setting panel would do that for you.  There was also a way to lock some values - if you picked M5, it didn't let you change width and height.  The code had grown old and ugly and was rewritten shortly before @PPAC37 and @coliss86 started contributing.  The plan is to build a list of machine profiles that can be picked from a list that makes it easier to run different styles of machines.

Currently width and height can be anything.  
Stepper wire, timing belt, and servo wire length are estimated lengths a user would need for a DIY machine (read only).
Diameter should be pen diameter
Max speed could be max travel speed (mm/s)
Draw speed could be max drawing speed (mm/s)
Acceleration is in mm/s/s
Up and Down could be Up angle and Down angle

Everything after that is hard coded values for advanced users and should probably be on a second tab (accessible but less obivous).  They are all Marlin hardcoded numbers that should match the firmware to get an accurate time estimate.

carl1961, 2022-02-07:
@PPAC37 @i-make-robots , Thanks for Your fast great answers, I understand.

Keepthefunk, 2022-02-08:
I would propose a direct servo speed option. So we can lower the servo speed from full speed down to something appropriate. This would reduce noise and pen tip wear. And the drawing accuracy could be highly increased because the gondola would shake less. 
If the pen could be lowered gently I wouldnt even see the need for a delay afterwards. Too long a wait could even cause ink bleeding at the start of each line.
Just my thoughts concerning servo movement.

i-make-robots, 2022-02-08:
There is no "direct servo speed".  servos go to their commanded angle as fast as they can.  Our solution is to send several intermediate steps spaced out over time to fake a slower speed.  if tuned badly the servo will stop and start many times, which might cause more shake.

Another solution might be to change the pen holder completely to use a [crank movement](http://507movements.com/mm_092.html) and a short wire from, say, a paper clip. the stroke length would be larger and the end of the stroke could be closer to when the servo arm is parallel with the wire - meaning it would move slower just before the pen touches/leaves the surface.

Keepthefunk, 2022-02-09:
Maybe the number of intermediate steps is just not sufficient. I hear two or three (Ssst-st-st).
I am dreaming of a smooth servo motion like in this video:
https://youtu.be/fC9uxdOBhfA?t=22 (the right servo, cubic)
Maybe even some more options for servo speed: [ ] No easing [  ] Quadratic easing or [x] cubic easing 
:-)

i-make-robots, 2022-02-09:
Discord user LordSox would be super interested in that.  The servo library in Marlin is the root of the problem, you'd have to talk to them about how to improve it.