MarginallyClever/Makelangelo-software #532: Reorder should be enabled by default

coliss86, 2022-02-08:
**Is your feature request related to a problem? Please describe.**
As the feature "reorder" seems to be stable, why not applying it automatically when loading a file after applying the effect ?

**Describe the solution you'd like**
After loading a SVG or when closing the "Load file preview", this optimization is applied.

**Describe alternatives you've considered**
None

**Additional context**
None

coliss86, 2022-02-08:
Why would you not want it ?
I think is it useful for most of the people, so we can add a checkbox on the "Load file preview" activated by default which enables it. So if you don't want it, you uncheck the box.
Or it can be activated for SVG as a first step.

i-make-robots, 2022-02-08:
And it would run after everything, every time? working with large art like ganesh would be painful.

coliss86, 2022-02-09:
a threshold can be configured for svg : if the number of lines or instructions is lower than 10000 for example, reorder is applied.

coliss86, 2022-02-09:
I just tried the re-ordering on a svg with 351228 instructions : it took 5 min. I better understand your concern.

I would like something which remind me to do the reorder before launching the robot. I don't want a popup, maybe a banner can do the work.

i-make-robots, 2022-02-09:
Personally I like that it makes people join the discord and become part of the community.  Maybe a link could be added under Help > Instruction Manual that points to [mcr.dozuki.com sub page.](https://mcr.dozuki.com/c/Makelangelo_3_and_5_Guide)

coliss86, 2022-02-09:
Hu ? I don't see the link between re-ordering and your answer ;)
What am I missing ?

i-make-robots, 2022-02-09:
When people find the program is slow, they go for help.  they find the discord.  they meet new friends.  they get a helpful answer quickly and are happy.  the community grows.  this is a small friction that creates positive benefit.

coliss86, 2022-02-10:
Ah good point !

closed 2022-02-10 by coliss86