MarginallyClever/Makelangelo-software #534: Load file Scratch 3

PPAC37, 2022-02-09:
**Steps to make the bug happen**
Try to open a Scratch 3 project save as a local file ( extension .sb3 ).
https://scratch.mit.edu/projects/641318749

**What was supposed to happen**
A clear and concise description of what you expected to happen.
* The Scratch file is open and render

A clear and concise description of what happens instead.
* Dialog error message whith Exceptions ...

 ( after changing the TurtleFactory isValidExtension 
https://github.com/MarginallyClever/Makelangelo-software/blob/ab4bf229e44eb53e146fd547569e607ceb83a59d/src/main/java/com/marginallyclever/makelangelo/makeArt/io/vector/TurtleFactory.java#L43

by 
~~~
if(filename.toLowerCase().endsWith(e.toLowerCase())) return true;
~~~
 to accept lower case extension "sb3"
)

~~~
java.lang.ClassCastException: class org.json.JSONObject$Null cannot be cast to class java.lang.String (org.json.JSONObject$Null is in unnamed module of loader 'app'; java.lang.String is in module java.base of loader 'bootstrap')
	
~~~
 
**Platform (please complete the following information):**
 - Version: github master 7.31.0
 - OS: Linux Ubuntu 18.04, 
 ~~~
$ java --version
java 17 2021-09-14 LTS
Java(TM) SE Runtime Environment (build 17+35-LTS-2724)
Java HotSpot(TM) 64-Bit Server VM (build 17+35-LTS-2724, mixed mode, sharing)
~~~

Is LoadFile Scratch 3 experimental (or not fully yet implemented) and this is normal 
or is this my JRE ? or something else ?

i-make-robots, 2022-02-09:
SB3 is incomplete.  The format is MUCH more complicated than SB2.

i-make-robots, 2022-02-09:
Do you have a sample to work from?

i-make-robots, 2022-02-09:
[inkylinesplot.zip](https://github.com/MarginallyClever/Makelangelo-software/files/8035208/inkylinesplot.zip)
[image]
[image]

i-make-robots, 2022-02-09:
first problem, case sensitive file extensions in `TurtleFactory.isValidExtension()` fixed in my branch.

i-make-robots, 2022-02-09:
First plot success in eb88f443

i-make-robots, 2022-02-10:
It works with https://github.com/MarginallyClever/Makelangelo-software/files/8035208/inkylinesplot.zip
OP sample is much more complex.

PPAC37, 2022-02-10:
Some minor changes and rendering of motion blocks, added in #537 

The management/interpretation of procedure and procedure call in a Scratch project remains out of reach for the moment ... (and it's been a while since I played to make an interpreter that needs a stack of calls and variable state stack so I'm rusty in their implementations)

i-make-robots, 2022-02-10:
[image]
[image]

closed 2022-02-10 by i-make-robots

PPAC37, 2022-02-11:
I take my hat off to you! / Respect!
Well done!

PPAC37, 2022-02-11:
unfortunately, now inkylinesplot.sb3 gives me an exception when loading it.
~~~
2022-02-11 08:02:07,751 ERROR c.m.m.makeArt.io.LoadFilePanel - Failed to load /home/q6/0_mcM_test/scratch3/inkylinesplot.sb3 
org.json.JSONException: JSONObject["VARIABLE"] not found.
	at org.json.JSONObject.get(JSONObject.java:572)
	at com.marginallyclever.makelangelo.makeArt.io.vector.LoadScratch3.findInputInBlock(LoadScratch3.java:457)
	at com.marginallyclever.makelangelo.makeArt.io.vector.LoadScratch3.setVariableTo(LoadScratch3.java:385)
	at com.marginallyclever.makelangelo.makeArt.io.vector.LoadScratch3.parseScratchCode(LoadScratch3.java:241)
	at com.marginallyclever.makelangelo.makeArt.io.vector.LoadScratch3.readScratchInstructions(LoadScratch3.java:203)
~~~

Also, in the case of a "poorly" done/"not clean" scratch project, (for example, if we leave a variable lying around, 
[image]

https://scratch.mit.edu/projects/642635174

we have some cast Exceptions at

https://github.com/MarginallyClever/Makelangelo-software/blob/f1da7ada92cc6bc95d8711a1dabdc3ac7386af07/src/main/java/com/marginallyclever/makelangelo/makeArt/io/vector/LoadScratch3.java#L195

solution ? 
~~~
	Object blocks_get_h = blocks.get(k);
			if ( ! ( blocks_get_h instanceof JSONObject )){
				// // TODO somethinge is not what expected ... maybe a lost variable in the scratch projet ...
				logger.debug("not expected: {} instanceof::{}",blocks_get_h,blocks_get_h.getClass().toString());
			}else{
...

~~~
and same at
https://github.com/MarginallyClever/Makelangelo-software/blob/f1da7ada92cc6bc95d8711a1dabdc3ac7386af07/src/main/java/com/marginallyclever/makelangelo/makeArt/io/vector/LoadScratch3.java#L613

PPAC37, 2022-02-11:
Cant find VARIABLE :
[StrangeOrbit remix.zip](https://github.com/MarginallyClever/Makelangelo-software/files/8051300/StrangeOrbit.remix.zip)
[image]

A casting exception (due to a variable lost in the project), and also an infinite loop (yes it's normal that this creates an infinite loop when reading, but that tells me that the infinite loop should not be implemented , or just loop it once and put a warning ... and normally illustrate the 90° orientation difference between Scratch and turtle. :
[base_pen_axis_lost_var_and_infinit_loop.zip](https://github.com/MarginallyClever/Makelangelo-software/files/8051301/base_pen_axis_lost_var_and_infinit_loop.zip)
[image]

PPAC37, 2022-02-11:
And a starnge one ... input0 not found 
[TurtleArt - never ending flower remix copy.zip](https://github.com/MarginallyClever/Makelangelo-software/files/8051390/TurtleArt.-.never.ending.flower.remix.copy.zip)
[image]

i-make-robots, 2022-02-11:
Latest update appears to pass all cases.  Thank you for the great puzzles!
new translation key `LoadScratch3.foreverNotAllowed` to warn users.  Easiest way for me was to just say no.

PPAC37, 2022-02-12:
[image]

Magnificent! I'm having fun with Scratch (working on my algorithmic)!
and Makelangelo gives me great results.

The essential scratch uses seem to be implemented.

I dare to ask
I have two features that will be user friendly but the last one is not without difficulty/risk.

1 - Color : As there is possibility in a Scratch project to define the color of the pen, it seems possible to take it into account in turtle to allow multi color (as in the graph paper generator).

2 - "parametric variables". As it is possible to define in the scratch monitor, variables with a Slider bar, to modify their value, a Dialoge/JPanel which will allow them to be modified (a bit like what is done with images or turtle generators, but there the SelectSlider would be dynamicaly built based on the visible variables, with a Slider bar, in the "monitor" of the scratch project. (this query seems like too much coding but eventually maybe, when we have settled most important things... :) )

PPAC37, 2022-02-12:
For colors if you have red/green 3D glasses a basic 3D anaglyph test
[image]

https://scratch.mit.edu/projects/643301613
[anaglyphe_000.zip](https://github.com/MarginallyClever/Makelangelo-software/files/8053436/anaglyphe_000.zip)

PPAC37, 2022-02-12:
operator modulo seem to be missing.
[image]
https://scratch.mit.edu/projects/643326693
[modulo_and_round.zip](https://github.com/MarginallyClever/Makelangelo-software/files/8053605/modulo_and_round.zip)

i-make-robots, 2022-02-12:
I haven’t found the color field yet. Could be done.
Sliders would mean a two step loading process, very complex from here.
Mod can happen.

PPAC37, 2022-02-12:
setAngle value in Turtle do not match scratch DIRECTION orientation ...
[image]

https://scratch.mit.edu/projects/643375721/
[orientation.zip](https://github.com/MarginallyClever/Makelangelo-software/files/8054075/orientation.zip)
[image]

i-make-robots, 2022-02-12:
I suspect the starting orientation is wrong.

commit f1da7ada (2022-02-10): fix #534 Load Scratch 3.0 Files

* clean imports
* fix case sensitive isValidExtension()
* remove redundant exceptions
* first success with inkylinkesplot.sb3
* continued...
* read and store procedure parameters
* #534 success with procedure calls
* javadoc
* cleaning LoadScratch2 based on LoadScratch3 lessons