MarginallyClever/Makelangelo-software #547: running two copies both write to same log file

i-make-robots, 2022-02-15:
**Steps to make the bug happen**
1. Run instance A
2. Load an image (seahorse.svg)
3. View log file.
4. Run instance B.
4. return to still-open log file.  startup B is in log file A.

**What was supposed to happen**
separate log file for each instance?

**Platform (please complete the following information):**
 - Version: 7.31.0
 - OS: Windows

**Log file**
[makelangelo.log](https://github.com/MarginallyClever/Makelangelo-software/files/8073832/makelangelo.log)

coliss86, 2022-02-15:
hum, quite difficult to solve :'( 
If you want 2 instances, how do you want to purge old log files ? which criteria do you use to select the correct log file of the previous instance ?

PPAC37, 2022-02-16:
yes it's not simple, it would be necessary to set up a mechanism that allows an instance to know its instantiation number. (a kind of singleton pattern)
But with java it's a nice headache if you want it functional for instances created by two JREs.

if I remember correctly, it requires using a system object like a socket (each instance tries to open it as a server and listens and if already exists (server) then if connects as a client and sends a message to say I am a new instance what and my instance number...) or the existence of a file (deleted at exit ...)

i-make-robots, 2022-02-16:
This will be closed and marked don’t fix. We are aware of it and we don’t plan to do anything. Who runs two copies at once?!

closed 2022-02-16 by i-make-robots