MarginallyClever/Makelangelo-software #560: Verbose must no be handle by a boolean

coliss86, 2022-02-19:
@PPAC37 in https://github.com/MarginallyClever/Makelangelo-software/pull/551/commits/12200cdcb39c0517aa70f71a6a8cfd3bda6eb235, there is a new boolean to make the class more verbose or not.
As [logback](https://logback.qos.ch/) is used in this application, this boolean is not needed.
You can silent or not the class by adding a line in https://github.com/MarginallyClever/Makelangelo-software/blob/master/src/main/resources/logback.xml#L36 like: 
```
<logger name="com.marginallyclever.makelangelo.makeArt.io.vector.LoadScratch3" level="debug" />
```
the `level` can be error, warn, info, debug, trace.

There is already some class that are less verbose than other: https://github.com/MarginallyClever/Makelangelo-software/blob/0396a8498125a1dbe67355450c00ab2d01b73780/src/main/resources/logback.xml#L15

If you think debug is to verbose, you can replace `logger.debug()` by `logger.trace()` and adjust the level of log in the config file accordingly.

PPAC37, 2022-02-19:
Thank you. 
I was reinventing the wheel when it already existed. 
I correct this as soon as possible.

PPAC37, 2022-02-19:
I need to study how i can change the level of a logger during execution (to facilitate my development/test, I have a main which launches LoadScratch3 on a set of test files. and just for him, I want to be able to adjust the logger level of LoadScratch3. )

coliss86, 2022-02-19:
just modify the logback file and don't commit it :-)  no need to adjust at runtime :-)

closed 2022-02-20 by coliss86