MarginallyClever/Makelangelo-software #562: File cleanup ?

coliss86, 2022-02-19:
**Is your feature request related to a problem? Please describe.**
I think some files are not needed anymore:
* Makelangelo.doxyfile
* other-resources/favicon.xcf
* src/main/java/com/marginallyclever/convenience/GridBagConstraintsLabel.java
* src/main/java/com/marginallyclever/convenience/GridBagConstraintsValue.java
* src/main/java/com/marginallyclever/convenience/Profiler.java
* src/main/java/com/marginallyclever/convenience/SpringUtilities.java
* src/main/java/com/marginallyclever/util/package-info.java

**Describe the solution you'd like**
I would like to delete them

PPAC37, 2022-02-19:
Also 

(double ?)
* [ ] https://github.com/MarginallyClever/Makelangelo-software/blob/master/src/main/art/makelangelo5-motors.png
* [ ] https://github.com/MarginallyClever/Makelangelo-software/blob/master/src/main/art/makelangelo5.png
(? may be better to store it elsewhere (not in src) )
https://github.com/MarginallyClever/Makelangelo-software/blob/master/src/main/art/makelangelo5.xcf

(some files (*.sb2, *.svg ) don't seem to be used in any test (but to be checked) maybe better to store them somewhere else?)
https://github.com/MarginallyClever/Makelangelo-software/tree/master/src/test/resources

coliss86, 2022-02-19:
no, keep src/test/resources, I'm planning to add unit test with those files

i-make-robots, 2022-02-19:
I have no issue with removing the suggested files.  Only keep what @coliss86 wants.

i-make-robots, 2022-02-20:
there shouldn't be an art folder.  if there is, it's probably meant to be the original source material and belongs in other-resources, which should be .ignored.

PR #566 (2022-02-21): 🔥 remove unused files
Fix #562
closed 2022-02-21 by i-make-robots