MarginallyClever/Makelangelo-software #564: Bad correction Scratch3 pointInDirection

PPAC37, 2022-02-20:
Sorry, I was wrong.
The current implementation of Scratch3 pointInDirection does not have a -90° difference with Turtle.setAngle.

It's rather (only for pointInDirection ) an axial symmetry of an axis having an angle of 45°... and there I don't know if it's Turtle's setAngle or if it's really that...

(if I make the correction in LoadScratch ...) I have trouble seeing clearly to apply this transformation in an elegant way to an orientation in °

https://scratch.mit.edu/projects/647531000

[debug_orientation_c_t_s_ap_d_fus.zip](https://github.com/MarginallyClever/Makelangelo-software/files/8103721/debug_orientation_c_t_s_ap_d_fus.zip)

Rendered under scratch
[image]

Rendered under Makelangelo (master) via LoadScratch3 
[image]

(n.b.: don't pay attention to the colors, I'm trying to implement pen_changePenColorParamTo and pen_changePenColorParamBy and I still have small adjustments in my calculations between RGB color and HSB color...)

PPAC37, 2022-02-20:
[image]

coliss86, 2022-02-20:
@PPAC37 can you have a look at the PR #567 which add unit test for scratch files.
This PR is based on the work on #563 and show duplicates code between the both, as soon as #563 is merged, I'll update #567.
When you'll make your fix, you should add a new test in LoadScratch3Test.java to make sure the expected behaviour does not change over time

PPAC37, 2022-02-20:
I'm really not fresh today, it took me all this time to finish with `myTurtle.setAngle(90.0-v);` instead of `myTurtle.setAngle(v-90.0);`

PR #571 (2022-02-22): fix_bad_fix point in direction (and some comments)
Close: #564
closed 2022-02-22 by i-make-robots