MarginallyClever/Makelangelo-software #568: SVG ellipse not interpreted correctly

coliss86, 2022-02-20:
**Steps to make the bug happen**
1. Load the attached svg [ellipse.svg](https://user-images.githubusercontent.com/23615562/154847077-b1173050-e782-4df0-b16b-344cf3de5328.svg)
2. the app convert it to a triangle : 
[image]
 

**What was supposed to happen**
It is a ellipse
[image]

**Platform (please complete the following information):**
 - Version: master
 - OS: macOS

**Log file**
```
2022-02-20 15:13:09,886 DEBUG c.m.makelangelo.Translator - starting translator... 
2022-02-20 15:13:09,900 DEBUG c.m.makelangelo.Translator - Default language = English 
2022-02-20 15:13:09,984 DEBUG c.m.makelangelo.Translator - loadConfig: User Preference Node: /DrawBot/Language=English 
2022-02-20 15:13:10,465 DEBUG c.m.makelangelo.paper.Paper - loadConfig() before Paper Width = 420.00 Height = 594.00 (left=-210.00 right=210.00 top=297.00 bottom=-297.00) centerShift(X= 0.00 Y= 0.00) color (255,255,255) 
2022-02-20 15:13:10,467 DEBUG c.m.makelangelo.paper.Paper - loadConfig() after Paper Width = 210.00 Height = 297.00 (left=-105.00 right=105.00 top=148.50 bottom=-148.50) centerShift(X= 0.00 Y= 0.00) color (255,255,255) 
2022-02-20 15:13:11,698 DEBUG c.m.util.PropertiesFileHelper - version: 7.31.0 
2022-02-20 15:13:11,699 DEBUG c.m.util.PropertiesFileHelper - fullGitRevision: dirty 
2022-02-20 15:13:11,726 DEBUG c.m.makelangelo.RecentFiles - loading recent files 
2022-02-20 15:13:13,088 DEBUG c.m.m.preview.PreviewPanel -   get GL capabilities... 
2022-02-20 15:13:13,092 DEBUG c.m.m.preview.PreviewPanel -   starting animator... 
2022-02-20 15:13:19,865 DEBUG c.m.m.makeArt.io.OpenFileChooser - File selected by user: /Users/guillaume/triangle.svg 
2022-02-20 15:13:19,949 DEBUG c.m.m.makeArt.io.vector.LoadSVG - Loading... 
2022-02-20 15:13:20,103 DEBUG c.m.m.makeArt.io.vector.LoadSVG - ...parse path 
2022-02-20 15:13:20,104 DEBUG c.m.m.makeArt.io.vector.LoadSVG - 0 elements 
2022-02-20 15:13:20,104 DEBUG c.m.m.makeArt.io.vector.LoadSVG - ...parse polylines 
2022-02-20 15:13:20,104 DEBUG c.m.m.makeArt.io.vector.LoadSVG - 0 elements 
2022-02-20 15:13:20,104 DEBUG c.m.m.makeArt.io.vector.LoadSVG - ...parse polygons 
2022-02-20 15:13:20,104 DEBUG c.m.m.makeArt.io.vector.LoadSVG - 0 elements 
2022-02-20 15:13:20,104 DEBUG c.m.m.makeArt.io.vector.LoadSVG - ...parse lines 
2022-02-20 15:13:20,104 DEBUG c.m.m.makeArt.io.vector.LoadSVG - 0 elements 
2022-02-20 15:13:20,104 DEBUG c.m.m.makeArt.io.vector.LoadSVG - ...parse rects 
2022-02-20 15:13:20,105 DEBUG c.m.m.makeArt.io.vector.LoadSVG - 0 elements 
2022-02-20 15:13:20,105 DEBUG c.m.m.makeArt.io.vector.LoadSVG - ...parse circles 
2022-02-20 15:13:20,105 DEBUG c.m.m.makeArt.io.vector.LoadSVG - 0 elements 
2022-02-20 15:13:20,105 DEBUG c.m.m.makeArt.io.vector.LoadSVG - ...parse ellipses 
2022-02-20 15:13:20,105 DEBUG c.m.m.makeArt.io.vector.LoadSVG - 1 elements 
```

PPAC37, 2022-02-20:
I think I have the reason for the defect.

~~~
logger.debug("circ = min(3,{})",Math.floor(Math.PI * ry*rx));
~~~
+
https://github.com/MarginallyClever/Makelangelo-software/blob/046682f054870411f4b554a18312204f33fe0823/src/main/java/com/marginallyclever/makelangelo/makeArt/io/vector/LoadSVG.java#L297-L304
=
~~~
2022-02-20 23:46:21,768 DEBUG c.m.m.makeArt.io.vector.LoadSVG - circ = min(3,3831.0) 
~~~

3 points ...
So with like 72 instead of 3 it gives
[image]

But I put 72 at random, it should surely be better chosen ...

i-make-robots, 2022-02-20:
Finding the right number of points for this was annoying.  Too many and it slows the plotter and eats ram for nothing.  too few and it looks like a triangle.  but also [getting a good number is hard](https://www.mathsisfun.com/geometry/ellipse-perimeter.html).  how about?
```java
double perimeterOfAnEllipseApprox = Math.PI * 2.0 * Math.sqrt((ry*ry + rx*rx)/2.0);
double steps = Math.max(3,perimeterOfAnEllipseApprox);
steps = Math.min(60,steps);
for(double i=1;i<perimeterOfAnEllipseApprox;++i) {
	double v = (Math.PI*2.0) * (i/perimeterOfAnEllipseApprox);
```

i-make-robots, 2022-02-20:
#572

PPAC37, 2022-02-20:
I tell myself that it is necessary to apply to perimeterOfAnEllipseApprox the transformation matrix to have a value more adapted according to the "zoom" (if the transformation matrix embeds the notion of zoom ...)

otherwise, this reminds me of the variable $fn of openscad to define the number of segments of an arc... there may be things to remember from their doc
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features

closed 2022-02-23 by coliss86