MarginallyClever/Makelangelo-software #578: LoadSVG, an approximation for CubicCurve, generateCurvePoints(...);

PPAC37, 2022-02-23:
**Is your feature request related to a problem? Please describe.**
Resolution.
I know it's a tricky compromise to find. And that it depends on the "fineness" of the original SVG and therefore probably not suitable for other SVGs which do not have the same "fineness".

**Describe the solution you'd like**
Find a way to have the ideal resolution in any case ...

**Describe alternatives you've considered**
Maybe to avoid complicating, a parameter that can be overloaded/redefined when launching Makelangelo via command line argument?

**Additional context**

https://github.com/MarginallyClever/Makelangelo-software/blob/094ec4a87595787d584735dd32a56112701a648a/src/main/java/com/marginallyclever/makelangelo/makeArt/io/vector/LoadSVG.java#L374

from 0.1 to 0.01 to 0.001
[image]

[image]
source https://freesvg.org/

PPAC37, 2022-02-23:
well this is definitely going to be a headache...
So we will avoid touching it at 0.1 because at 0.001 I get
[image]
[image]

i-make-robots, 2022-03-31:
I wonder if the SVG initially is very small.  this would mean a very small number of points in a curve.  then the SVG is scaled up to fill the paper and the problem becomes obvious.  If it was the other way around - load the SVG huge and then bring it down - there would be too many points in each curve.  Can't win!

closed 2022-06-10 by github-actions[bot]