MarginallyClever/Makelangelo-software #580: user personal "start/end gcode" "tool change start/end gcode"

PPAC37, 2022-02-24:
As in moderne 3D Slicers, a way to the user to define some personal GCode for a "plotter profile"
(no placeholder for the beginning)

coliss86, 2022-02-24:
Love this !
I started a discussion about it in https://github.com/MarginallyClever/Makelangelo-software/issues/552 but I did not implemented it.

PPAC37, 2022-02-24:
I have not yet passed the stage of the idea (feasibility analysis) , 
because the accumulation of 
* the MCV, and 
* the preferences usage, and 
* the PlotterSettingsPanel is already well filled and 
* the fact of making this usable also by the plotterinterface 

which makes the task of implementing this complicated ... 

(in addition I would have liked placeholder / replacement pattern ... like in recent 3D slicer ex: http://files.fieldofview.com/cura/Replacement_Patterns.html )

i-make-robots, 2022-02-27:
replacement patterns would only happen when saving.  In theory they could be implemented across all save files - so saving svgs could have a comment with info in them, too.

It sounds to me like a problem is being attacked from multiple directions without a plan.  Maybe we should discuss it first.

coliss86, 2022-02-27:
> replacement patterns would only happen when saving.

No, even the printing directly from the software in order to have the same behavior.

> It sounds to me like a problem is being attacked from multiple directions without a plan. Maybe we should discuss it first.

What do you think ? Have you start reading https://github.com/MarginallyClever/Makelangelo-software/pull/583 ?

coliss86, 2022-05-07:
I'm working on it

closed 2022-07-14 by github-actions[bot]