MarginallyClever/Makelangelo-software #585: Plotter settings Save or cancel do not close the dialog

PPAC37, 2022-02-25:
all in the title.

coliss86, 2022-02-27:
I would like to see a "reset" button instead of "cancel". maybe a third button "close" to close the window

i-make-robots, 2022-02-27:
I hate extra buttons.  the [x] button is enough.  We can make the dialog non-resizable?  `setResizable(false);` ?

i-make-robots, 2022-03-08:
More thoughts on this...   like Prusa slicer, there should be a table of profiles.  a profile is a specfic set of plotter settings.

- [ ] with defaults for some machines (m5, m3.3, huge, etc).
- [ ] user can choose a profile and 
- [ ] then save it as a new design and tweak it.  
- [ ] user can't delete the default profiles.  
- [ ] user CAN delete custom profiles.
- [ ] all profiles can be saved to disk for unit testing/backups.
- [ ] user can swap between profiles non-destructively.
- [ ] plotter rendered in preview controlled by profile.

i-make-robots, 2022-03-08:
also overlaps with #603

PR #610 (2022-03-11): 💄 prevent the PlotterSettingsPanel to be resizeable
Fix #585
closed 2022-03-11 by i-make-robots