MarginallyClever/Makelangelo-software #587: a color value association to color name does not match w3c color name for green...

PPAC37, 2022-02-26:
if we want to name the color as in https://www.w3.org/TR/css-color-3/#normative

https://github.com/MarginallyClever/Makelangelo-software/blob/33bc26eedbb22490785d33ac550ab710d3fb1a56/src/main/java/com/marginallyclever/makelangelo/plotter/plotterControls/MarlinPlotterInterface.java#L193-L201

~~~
		case 0x00ff00:// = w3c "lime" https://www.w3.org/TR/css-color-3/#normative
			name = "lime";
			break;
		case 0x008000:// = w3c "green" https://www.w3.org/TR/css-color-3/#normative
			name = "green";
			break;
~~~

yes it's not a big deal but ...

i-make-robots, 2022-02-26:
It would be a good idea to have a matrix? Tree? of styles and colors. A style might be “comic” and then the sun elements are colic marker designations and their hex code. Another would be pantone. Etc.

PPAC37, 2022-02-26:
You think of color sets that match some theme ?
like "christmas" = { "red", "green", "white", ... }
and 
[image]

Or a way to present / group the colors that would be close to each other in a color chart ?
( a bit like what is done on https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Transformations#color )

i-make-robots, 2022-02-26:
No... I was thinking more like official color codes from pen makers.  [copic](https://copic.jp/en/about/color-system/), staedler (can't find), and (pantone)[https://www.pantone-colours.com/].

PPAC37, 2022-02-26:
maybe there is a way around this, like making a voluntary typo so that it is not exactly the same name but that it is easily guessed like Bleu => BeBleu , Red => ReRed? ;)

coliss86, 2022-02-27:
why not just the name as defined in the w3c ? no copyright problem and names on the list are already chosen...

coliss86, 2022-02-27:
like this: https://www.w3.org/wiki/CSS/Properties/color/keywords

PPAC37, 2022-02-27:
Yes to use W3C color naming.
It will just be necessary to agree on when there are several names for the same value which one is used.

ex : 
~~~

 case 0x00ffff: name = "aqua"; // [aqua, cyan]
break;

 case 0xff00ff: name = "fuchsia"; // [fuchsia, magenta]
break;

 case 0x808080: name = "gray"; // [gray, grey]
break;

 case 0x2f4f4f: name = "darkslategray"; // [darkslategray, darkslategrey]
break;
 case 0x696969: name = "dimgray"; // [dimgray, dimgrey]
break;
 case 0x708090: name = "slategray"; // [slategray, slategrey]
break;
 case 0x778899: name = "lightslategray"; // [lightslategray, lightslategrey]
break;
 case 0xa9a9a9: name = "darkgray"; // [darkgray, darkgrey]
break;
 case 0xd3d3d3: name = "lightgray"; // [lightgray, lightgrey]
break;

~~~

i-make-robots, 2022-04-27:
f4070461 introduced com.marginallyclever.convenience.SVGColorNames which contains all the W3C basic and extended named colors.  I will use this to find the names of each color instead of the switch above.  addressed in #655

closed 2022-05-09 by i-make-robots