MarginallyClever/Makelangelo-software #601: too much pen up / pen down

coliss86, 2022-03-05:
**Steps to make the bug happen**
1. Load the attached svg
2. simplify
3. reorder
4. print it

here is the svg : [reinder-torus-90e6288a6b(1)](https://user-images.githubusercontent.com/23615562/156883994-a634803d-8973-46f4-8289-ad7e2258b8a3.svg)

**What was supposed to happen**
At the beginning, the pen is moving up and down and draw very small part

**Platform (please complete the following information):**
 - Version: master
 - OS: macos
 - Robot model (if any): [e.g. Makelangelo 5]

**Log file**
```
2022-03-05 13:50:57,623 DEBUG c.m.makelangelo.Translator - starting translator... 
2022-03-05 13:50:57,632 DEBUG c.m.makelangelo.Translator - Default language = English 
2022-03-05 13:50:57,707 DEBUG c.m.makelangelo.Translator - loadConfig: User Preference Node: /DrawBot/Language=English 
2022-03-05 13:50:58,027 DEBUG c.m.makelangelo.paper.Paper - loadConfig() before Paper Width = 420.00 Height = 594.00 (left=-210.00 right=210.00 top=297.00 bottom=-297.00) centerShift(X= 0.00 Y= 0.00) color (255,255,255) 
2022-03-05 13:50:58,028 DEBUG c.m.makelangelo.paper.Paper - loadConfig() after Paper Width = 210.00 Height = 297.00 (left=-105.00 right=105.00 top=148.50 bottom=-148.50) centerShift(X= 0.00 Y= 0.00) color (255,255,255) 
2022-03-05 13:50:59,176 DEBUG c.m.util.PropertiesFileHelper - fullGitRevision: title-452e77e 
2022-03-05 13:50:59,176 DEBUG c.m.util.PropertiesFileHelper - version: 7.32.0 
2022-03-05 13:50:59,217 DEBUG c.m.makelangelo.RecentFiles - loading recent files 
2022-03-05 13:51:00,530 DEBUG c.m.m.preview.PreviewPanel -   get GL capabilities... 
2022-03-05 13:51:00,534 DEBUG c.m.m.preview.PreviewPanel -   starting animator... 
2022-03-05 13:51:05,943 DEBUG c.m.m.makeArt.io.vector.LoadSVG - Loading... 
2022-03-05 13:51:06,236 DEBUG c.m.m.makeArt.io.vector.LoadSVG - ...parse path 
2022-03-05 13:51:06,238 DEBUG c.m.m.makeArt.io.vector.LoadSVG - 0 elements 
2022-03-05 13:51:06,238 DEBUG c.m.m.makeArt.io.vector.LoadSVG - ...parse polylines 
2022-03-05 13:51:06,238 DEBUG c.m.m.makeArt.io.vector.LoadSVG - 2192 elements 
2022-03-05 13:51:06,250 DEBUG c.m.m.makeArt.io.vector.LoadSVG - ...parse polygons 
2022-03-05 13:51:06,251 DEBUG c.m.m.makeArt.io.vector.LoadSVG - 0 elements 
2022-03-05 13:51:06,251 DEBUG c.m.m.makeArt.io.vector.LoadSVG - ...parse lines 
2022-03-05 13:51:06,251 DEBUG c.m.m.makeArt.io.vector.LoadSVG - 0 elements 
2022-03-05 13:51:06,251 DEBUG c.m.m.makeArt.io.vector.LoadSVG - ...parse rects 
2022-03-05 13:51:06,251 DEBUG c.m.m.makeArt.io.vector.LoadSVG - 0 elements 
2022-03-05 13:51:06,251 DEBUG c.m.m.makeArt.io.vector.LoadSVG - ...parse circles 
2022-03-05 13:51:06,252 DEBUG c.m.m.makeArt.io.vector.LoadSVG - 0 elements 
2022-03-05 13:51:06,252 DEBUG c.m.m.makeArt.io.vector.LoadSVG - ...parse ellipses 
2022-03-05 13:51:06,252 DEBUG c.m.m.makeArt.io.vector.LoadSVG - 0 elements 
2022-03-05 13:51:12,206 DEBUG c.m.m.makeArt.SimplifyTurtleAction - begin @ 4386 
2022-03-05 13:51:12,207 DEBUG c.m.makelangelo.turtle.Turtle -   Found 4386 instructions. 
2022-03-05 13:51:12,209 DEBUG c.m.m.makeArt.SimplifyTurtleAction -   Converted to 2192 lines. 
2022-03-05 13:51:12,212 DEBUG c.m.m.makeArt.SimplifyTurtleAction -   - 0 shorts = 2192 lines. 
2022-03-05 13:51:12,213 DEBUG c.m.m.makeArt.SimplifyTurtleAction - end @ 4378 
2022-03-05 13:51:16,559 DEBUG c.m.m.makeArt.ReorderTurtleAction - reorder() start @ 4378 instructions. 
2022-03-05 13:51:16,560 DEBUG c.m.makelangelo.turtle.Turtle - Turtle.splitByToolChange() into 2 sections. 
2022-03-05 13:51:16,560 DEBUG c.m.makelangelo.turtle.Turtle - Turtle.splitByToolChange() 1 not-empty sections. 
2022-03-05 13:51:16,560 DEBUG c.m.m.makeArt.ReorderTurtleAction - reorder() layers: 1 
2022-03-05 13:51:16,560 DEBUG c.m.makelangelo.turtle.Turtle -   Found 4378 instructions. 
2022-03-05 13:51:16,561 DEBUG c.m.m.makeArt.ReorderTurtleAction -   (0,0,0) converted to 2193 lines. 
2022-03-05 13:51:16,561 DEBUG c.m.m.makeArt.ReorderTurtleAction -   greedyReordering() 
2022-03-05 13:51:16,597 DEBUG c.m.m.makeArt.ReorderTurtleAction - reorder() end @ 2356 instructions. 
2022-03-05 13:51:18,968 DEBUG c.m.m.p.p.ChooseConnection - Fetching connections 
2022-03-05 13:51:18,970 DEBUG c.m.c.serial.SerialTransportLayer - Listing available serial port 
2022-03-05 13:51:19,125 DEBUG c.m.c.serial.SerialTransportLayer -   /dev/cu.Bluetooth-Incoming-Port 
2022-03-05 13:51:19,125 DEBUG c.m.c.serial.SerialTransportLayer -   /dev/cu.Drawbot-Port 
2022-03-05 13:51:19,125 DEBUG c.m.c.serial.SerialTransportLayer -   /dev/cu.SRS-XB33-BluetoothSeria 
2022-03-05 13:51:19,125 DEBUG c.m.c.serial.SerialTransportLayer -   /dev/cu.robot-SerialPort 
2022-03-05 13:51:43,553 DEBUG c.m.m.p.p.MarlinInterface - MarlinInterface connected. 
2022-03-05 13:51:53,559 TRACE c.m.m.p.p.MarlinInterface - Heartbeat: M400 
2022-03-05 13:51:53,560 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 13:51:59,396 TRACE c.m.m.p.p.MarlinInterface - sending 'N1 M280 P0 S90 T50*99' 
2022-03-05 13:51:59,396 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 13:51:59,396 TRACE c.m.m.p.p.MarlinInterface - sending 'N2 G28 XY*48' 
2022-03-05 13:51:59,396 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 13:52:03,560 TRACE c.m.m.p.p.MarlinInterface - Heartbeat: M400 
[... 44 lines cut ...]
2022-03-05 13:52:11,961 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 13:52:11,961 TRACE c.m.m.p.p.MarlinInterface - sending 'N25 G1 X41.315 Y-48.653 F3000.0*70' 
2022-03-05 13:52:11,961 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 13:52:11,961 TRACE c.m.m.p.p.MarlinInterface - sending 'N26 G1 X37.958 Y-45.392 F3000.0*66' 
2022-03-05 13:52:11,961 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 13:52:11,962 TRACE c.m.m.p.p.MarlinInterface - sending 'N27 G1 X34.870 Y-42.350 F3000.0*66' 
2022-03-05 13:52:11,962 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 13:52:11,962 TRACE c.m.m.p.p.MarlinInterface - sending 'N28 G1 X32.024 Y-39.482 F3000.0*70' 
2022-03-05 13:52:11,962 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 13:52:11,962 TRACE c.m.m.p.p.MarlinInterface - sending 'N29 G1 X29.398 Y-36.752 F3000.0*72' 
2022-03-05 13:52:11,963 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 13:52:11,963 TRACE c.m.m.p.p.MarlinInterface - sending 'N30 G1 X26.973 Y-34.128 F3000.0*73' 
2022-03-05 13:52:11,963 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 13:52:11,963 TRACE c.m.m.p.p.MarlinInterface - sending 'N31 G1 X25.825 Y-32.823 F3000.0*77' 
2022-03-05 13:52:11,963 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 13:52:11,963 TRACE c.m.m.p.p.MarlinInterface - sending 'N32 M280 P0 S90 T50*83' 
2022-03-05 13:52:11,963 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 13:52:11,964 TRACE c.m.m.p.p.MarlinInterface - sending 'N33 G0 X26.176 Y-33.221*48' 
2022-03-05 13:52:11,964 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 13:52:11,965 TRACE c.m.m.p.p.MarlinInterface - sending 'N34 M280 P0 S25 T50*91' 
2022-03-05 13:52:11,965 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 13:52:11,965 TRACE c.m.m.p.p.MarlinInterface - sending 'N35 G1 X26.594 Y-33.277 F3000.0*71' 
2022-03-05 13:52:11,965 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
....
```

i-make-robots, 2022-03-07:
I don't see the svg - the image is white on white?

i-make-robots, 2022-03-07:
huh.  it loaded with lines OK in Makelangelo software.

i-make-robots, 2022-04-01:
[image]
https://github.com/MarginallyClever/Makelangelo-software/pull/642

closed 2022-04-02 by i-make-robots