MarginallyClever/Makelangelo-software #602: Robot stuck

coliss86, 2022-03-05:
**Steps to make the bug happen**
1. draw something with the robot connected with an USB cable
2. sometimes, it get stuck and I don't understand why.

**What was supposed to happen**
The robot should recover the error and continue

**Platform (please complete the following information):**
 - Version: master
 - OS: macos
 - Robot model (if any): DYI

**Log file**
Here is the log :
```
2022-03-05 14:03:54,529 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 14:03:54,530 TRACE c.m.m.p.p.MarlinInterface - sending 'N176 G1 X46.804 Y-57.418 F3000.0*126' 
2022-03-05 14:03:54,531 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 14:03:54,531 TRACE c.m.m.p.p.MarlinInterface - sending 'N177 G1 X50.931 Y-62.207 F3000.0*113' 
2022-03-05 14:03:55,344 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 14:03:55,344 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 14:03:55,344 TRACE c.m.m.p.p.MarlinInterface - sending 'N178 G1 X55.502 Y-67.607 F3000.0*118' 
2022-03-05 14:03:55,345 TRACE c.m.m.p.p.MarlinInterface - sending 'N179 G1 X60.607 Y-73.785 F3000.0*121' 
2022-03-05 14:03:55,347 TRACE c.m.m.p.p.MarlinInterface - received 'Error:checksum mismatch, Last Line: 174' 
2022-03-05 14:03:55,347 ERROR c.m.m.p.p.MarlinInterface - Error from printer 'Error:checksum mismatch, Last Line: 174' 
2022-03-05 14:03:55,347 TRACE c.m.m.p.p.MarlinInterface - received 'Resend: 175' 
2022-03-05 14:03:55,347 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 14:03:55,347 TRACE c.m.m.p.p.MarlinInterface - sending 'N175 G1 X43.051 Y-53.118 F3000.0*113' 
2022-03-05 14:03:55,349 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 14:03:55,349 TRACE c.m.m.p.p.MarlinInterface - sending 'N176 G1 X46.804 Y-57.418 F3000.0*126' 
2022-03-05 14:03:55,389 TRACE c.m.m.p.p.MarlinInterface - received 'Error:Line Number is not Last Line Number+1, Last Line: 174' 
2022-03-05 14:03:55,389 ERROR c.m.m.p.p.MarlinInterface - Error from printer 'Error:Line Number is not Last Line Number+1, Last Line: 174' 
2022-03-05 14:03:55,389 TRACE c.m.m.p.p.MarlinInterface - received 'Resend: 175' 
2022-03-05 14:03:55,389 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 14:03:55,389 TRACE c.m.m.p.p.MarlinInterface - received 'echo:Unknown command: "0"' 
2022-03-05 14:03:55,389 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 14:03:55,390 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 14:03:55,390 TRACE c.m.m.p.p.MarlinInterface - sending 'N175 G1 X43.051 Y-53.118 F3000.0*113' 
2022-03-05 14:03:55,390 TRACE c.m.m.p.p.MarlinInterface - sending 'N176 G1 X46.804 Y-57.418 F3000.0*126' 
2022-03-05 14:03:55,390 TRACE c.m.m.p.p.MarlinInterface - sending 'N177 G1 X50.931 Y-62.207 F3000.0*113' 
2022-03-05 14:03:55,391 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 14:03:55,391 TRACE c.m.m.p.p.MarlinInterface - sending 'N178 G1 X55.502 Y-67.607 F3000.0*118' 
2022-03-05 14:03:55,392 TRACE c.m.m.p.p.MarlinInterface - sending 'N179 G1 X60.607 Y-73.785 F3000.0*121' 
2022-03-05 14:03:55,438 TRACE c.m.m.p.p.MarlinInterface - received 'Error:Line Number is not Last Line Number+1, Last Line: 176' 
2022-03-05 14:03:55,438 ERROR c.m.m.p.p.MarlinInterface - Error from printer 'Error:Line Number is not Last Line Number+1, Last Line: 176' 
2022-03-05 14:03:55,438 TRACE c.m.m.p.p.MarlinInterface - received 'Resend: 177' 
2022-03-05 14:03:55,438 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 14:03:55,438 TRACE c.m.m.p.p.MarlinInterface - sending 'N177 G1 X50.931 Y-62.207 F3000.0*113' 
2022-03-05 14:03:55,440 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 14:03:55,440 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 14:03:55,440 TRACE c.m.m.p.p.MarlinInterface - sending 'N178 G1 X55.502 Y-67.607 F3000.0*118' 
2022-03-05 14:03:55,440 TRACE c.m.m.p.p.MarlinInterface - sending 'N179 G1 X60.607 Y-73.785 F3000.0*121' 
2022-03-05 14:03:55,440 TRACE c.m.m.p.p.MarlinInterface - sending 'N180 G1 X66.369 Y-80.960 F3000.0*125' 
2022-03-05 14:03:55,495 TRACE c.m.m.p.p.MarlinInterface - received 'Error:Line Number is not Last Line Number+1, Last Line: 179' 
2022-03-05 14:03:55,495 ERROR c.m.m.p.p.MarlinInterface - Error from printer 'Error:Line Number is not Last Line Number+1, Last Line: 179' 
2022-03-05 14:03:55,495 TRACE c.m.m.p.p.MarlinInterface - received 'Resend: 180' 
2022-03-05 14:03:55,495 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 14:03:55,495 TRACE c.m.m.p.p.MarlinInterface - received 'echo:Unknown command: "118"' 
2022-03-05 14:03:55,495 TRACE c.m.m.p.p.MarlinInterface - received 'ok' 
2022-03-05 14:03:55,495 TRACE c.m.m.p.p.MarlinInterface - sending 'N180 G1 X66.369 Y-80.960 F3000.0*125' 
2022-03-05 14:03:55,495 TRACE c.m.m.p.p.MarlinInterface - sending 'N181 G1 X72.960 Y-89.439 F3000.0*114' 
2022-03-05 14:03:55,495 TRACE c.m.m.p.p.MarlinInterface - sending 'N182 G1 X76.755 Y-94.500 F3000.0*122' 
2022-03-05 14:03:55,495 TRACE c.m.m.p.p.MarlinInterface - sending 'N183 M280 P0 S90 T50*104' 
2022-03-05 14:03:55,495 TRACE c.m.m.p.p.MarlinInterface - sending 'N184 G0 X67.513 Y-94.500*6' 
```

Nothing happened after `N184`

i-make-robots, 2022-03-05:
Almost guaranteed to be silent usb disconnect failure. I’ve seen it many times.

coliss86, 2022-03-11:
From https://github.com/MarginallyClever/Makelangelo-software/pull/611#discussion_r824285977 : when an error is detected, close and reopen the connection, home the robot and restart the drawing

closed 2022-05-19 by github-actions[bot]