MarginallyClever/Makelangelo-software #614: Save gcode to one file per color

i-make-robots, 2022-03-14:
**Is your feature request related to a problem? Please describe.**
Running a CMYK drawing takes a very long time.  It would be good if users could rest easy knowing the machine will stop when it finish at the end of a color.

**Describe the solution you'd like**
filename-color.gcode should be the format, where color is the hex code without a leading '#' or '0x'.

**Describe alternatives you've considered**
Making the user go through other apps in a circuitous route to achieve the same thing.

**Additional context**
- It doesn't have to be just CMYK.  It could be any combination of colors.  
- For a single color it should not add a color name to the file.  
- This is a new behavior that does not change the old save mechanism.  It is entirely optional to the user.
- It must appear in PlotterControls 
- It would be nice to also appear in File > Save As...

Keepthefunk, 2022-03-14:
Dan, everyone, thanks for addressing this!

crisgodinez12, 2022-03-15:
From what I've experience thus far, if the colors can be separated in different dxf or svg files, then that would help with the allotted time I have to print a CMYK because the computers at my school turn off at a certain time even though I have a no sleep for the pc and screen saver. Another benefit to this feature would be that I can then plot those files on smaller machines like the Eggbot, iDraw, or AxiDraw. I would also love this feature available because the CMYK layers are easier and smaller file size than other CMYK processors as seen in Drawbot V3 or others.  

Also, not sure if why, but the last time I ran the CMYK plot in the Makelangelo software, I expected for the servo to be up whence the yellow layer was done, but the servo arm was down, so the marker soaked onto the paper. I anticipated for the servo arm to be up, so that I could then return to the machine later and switch markers. 

Let me know if these requests are possible.

i-make-robots, 2022-03-16:
drafted in 11fd8e2a

i-make-robots, 2022-03-17:
unit test is still incoming

closed 2022-03-17 by i-make-robots