MarginallyClever/Makelangelo-software #622: No more log

coliss86, 2022-03-21:
**Steps to make the bug happen**
1. Launch the app

**What was supposed to happen**
The log of the application must be written to stdout or in a log file, but nothing anymore

**Platform (please complete the following information):**
 - Version: master
 - OS: macOS

**Log file**
```
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
```

**Additional context**
I think this due the dependency NodeGraphCore which come with logback-classic in version 1.3.0-alpha14, it was 1.2.10 before adding this dependency

coliss86, 2022-03-21:
#623 was an attempt to solve the problem by reverting to a stable release of logback but it fails due to nodegraphcore requiring the non stable release which seems to either have a bug or I can't find a valid configuration.

@i-make-robots can you please revert the version of logback in nodegraphcore to 1.2.11 and delete the file `src/main/resources/logback.xml` as it interferes with makelangelo please ?

i-make-robots, 2022-03-22:
If I delete the logback.xml then nodegraphcore won't log.  There must be a way...

coliss86, 2022-03-22:
> If I delete the logback.xml then nodegraphcore won't log. There must be a way...

nodegraphcore can be run as a standalone application ? I thought it was exclusively a library.
If yes, the log you are saying are probably for unit test and you can move `logback.xml` to `src/test/resources/logbatck-test.xml`
If not, you'll have to exclude this file from the published jar with this option : https://maven.apache.org/plugins/maven-jar-plugin/examples/include-exclude.html

i-make-robots, 2022-03-24:
you mean like in the `<resources>` section?
```
        <resources>
            <resource>
                <directory>${project.basedir}/src/main/resources</directory>
                <filtering>true</filtering>
                <includes>
                    <include>**/*.properties</include>
                </includes>
            </resource>
            <resource>
                <directory>${project.basedir}/src/main/resources</directory>
                <filtering>false</filtering>
                <excludes>
                    <exclude>**/*.properties</exclude>
                </excludes>
            </resource>
        </resources>
```

i-make-robots, 2022-03-24:
ooooh.... use the resource when building, but ignore it in the Jar.  totally different?

coliss86, 2022-03-24:
The resources section at the root of the pom is used for building.
Make sure you read the doc of the plugin, it has the same syntax but the configuration is for the plugin

i-make-robots, 2022-03-25:
appears fixed in 076cfbad.  solution was to roll back logback.classic from `1.3.0-alpha14` to `1.2.10`.

closed 2022-03-25 by i-make-robots