MarginallyClever/Makelangelo-software #658: Infinite loop at the end of the drawing

coliss86, 2022-07-06:
**Steps to make the bug happen**
1. Setup an end gcode
1. Draw a picture

**What happened **
At the end of the drawing, the app is stuck in a infinite loop with the robot while sending over and over the same line. If I click on "Emergency Stop", it stops.
The marlin logs shows the following: 
[image]

`; Park position` is in the end user gcode.

**Platform (please complete the following information):**
 - Version: master
 - OS: macos
 - Robot model (if any): custom

PPAC37, 2022-07-06:
Hi, it seems to me that comments are not removed in the current implementation for start and end user g-code. However, in principle, the comments should not be sent to the marlin firmware (the firmware does not have the role of filtering them and because in the idea already that the serial connection is slow, we only pass the necessary .. )

coliss86, 2022-07-07:
Fixed in #658, thanks !