MarginallyClever/Makelangelo-software #671: Makelagelo software dmg not opening on OSX 13 Ventura

NicoMiceli, 2022-11-07:
**Steps to make the bug happen**

I'm on the new OSX version 13, and when I download and install the dmg, I get the error ""Makelangelo” is damaged and can’t be opened. You should move it to the Trash."

 [image]

I saw the workaround here https://mcr.dozuki.com/Guide/1.+How+to+use+Makelangelo+Software+7.32+/39?lang=en#s434
However, I do not see the ability to approve that file. 

See the new settings screen here right after I tried to open the makelangelo 7.32.0:
[image]

**What was supposed to happen**
I was expecting to see the error based don't the docs, but in the new version of OSX, I am not seeing the ability to approve the file. 

**Platform (please complete the following information):**
 - Version: 13.0 (22A380)
 - OS: OSX

i-make-robots, 2022-11-07:
How thorough!  I love it.

Does this work?

1. In the Finder on your Mac, locate the app you want to open.
2. Don’t use Launchpad to do this. Launchpad doesn’t allow you to access the shortcut menu.
3. Control-click the app icon, then choose Open from the shortcut menu.
4. Click Open.
5. The app is saved as an exception to your security settings, and you can open it in the future by double-clicking it just as you can any registered app.
-- https://support.apple.com/en-ca/guide/mac-help/mh40616/mac

NicoMiceli, 2022-11-08:
Thanks for the quick response! 

That didn't work by itself but I used that and a combination of this command: 
`sudo xattr -cr /Applications/Makelangelo.app` 
that I stumbled over in this ["old" documentation](https://mcr.dozuki.com/Guide/%5Bold%5D+How+to+use+Makelangelo+Software+7.24-7.31/5?lang=en), **and it worked!** Can that get added to the new docs [here](https://mcr.dozuki.com/Guide/1.+How+to+use+Makelangelo+Software+7.32+/39?lang=en#s434), it might help other mac folks. 

Cheers!

i-make-robots, 2022-11-08:
Done.  Do you consider this solved to your satisfaction?

NicoMiceli, 2022-11-12:
Definitely! Thank you.

closed 2022-11-12 by NicoMiceli