MarginallyClever/Makelangelo-software #677: Unable to run on Apple silicon Mac

jasonluther, 2023-01-30:
I am not able to run Makelangelo on my Apple Macbook Air M2 (Apple silicon), 2022, running Ventura 13.1 (22C65). Using a fresh checkout of master and building with `./mvnw clean install`, I run it with `java -jar target/Makelangelo-7.34.1-with-dependencies.jar` and get this message:

```
Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: /private/var/folders/_s/rwt5whxs2f11s2qzwz5pd97r0000gn/T/jogamp_0000/file_cache/jln12894535790123853005/jln5669326532083840778/natives/macosx-universal/libgluegen_rt.dylib: dlopen(/private/var/folders/_s/rwt5whxs2f11s2qzwz5pd97r0000gn/T/jogamp_0000/file_cache/jln12894535790123853005/jln5669326532083840778/natives/macosx-universal/libgluegen_rt.dylib, 0x0001): tried: '/private/var/folders/_s/rwt5whxs2f11s2qzwz5pd97r0000gn/T/jogamp_0000/file_cache/jln12894535790123853005/jln5669326532083840778/natives/macosx-universal/libgluegen_rt.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/private/var/folders/_s/rwt5whxs2f11s2qzwz5pd97r0000gn/T/jogamp_0000/file_cache/jln12894535790123853005/jln5669326532083840778/natives/macosx-universal/libgluegen_rt.dylib' (no such file), '/private/var/folders/_s/rwt5whxs2f11s2qzwz5pd97r0000gn/T/jogamp_0000/file_cache/jln12894535790123853005/jln5669326532083840778/natives/macosx-universal/libgluegen_rt.dylib
```

These warnings during the build seem notable:
```
[INFO] --- maven-assembly-plugin:3.3.0:single (make-assembly) @ Makelangelo ---
[INFO] Reading assembly descriptor: /Users/jason/Documents/git/Makelangelo-software/src/main/assembly/application.xml
[WARNING] The following patterns were never triggered in this artifact inclusion filter:
o  'org.jogamp.gluegen:gluegen-rt:jar:natives-linux-amd64'
o  'org.jogamp.jogl:jogl-all:jar:natives-linux-amd64'

[WARNING] The following patterns were never triggered in this artifact inclusion filter:
o  'org.jogamp.gluegen:gluegen-rt:jar:natives-linux-i586'
o  'org.jogamp.jogl:jogl-all:jar:natives-linux-i586'

[WARNING] The following patterns were never triggered in this artifact inclusion filter:
o  'org.jogamp.gluegen:gluegen-rt:jar:natives-windows-amd64'
o  'org.jogamp.jogl:jogl-all:jar:natives-windows-amd64'

[WARNING] The following patterns were never triggered in this artifact inclusion filter:
o  'org.jogamp.gluegen:gluegen-rt:jar:natives-windows-i586'
o  'org.jogamp.jogl:jogl-all:jar:natives-windows-i586'

[WARNING] The following patterns were never triggered in this artifact inclusion filter:
o  'org.jogamp.gluegen:gluegen-rt:jar:natives-macosx-universal'
o  'org.jogamp.jogl:jogl-all:jar:natives-macosx-universal'
```

I have tried with different JDKs with the same result. Following the same steps, Makelangelo runs on an Intel Mac running Ventura 13.1. 

Build and run logs attached:
[mvnw-log.txt](https://github.com/MarginallyClever/Makelangelo-software/files/10538528/mvnw-log.txt)
[runlog.txt](https://github.com/MarginallyClever/Makelangelo-software/files/10538529/runlog.txt)

I will also note that I'm not able to run the released binary from the .dmg at https://github.com/MarginallyClever/Makelangelo-software/releases/tag/v7.32.0, despite running `sudo xattr -cr /Applications/Makelangelo.app` as suggested in https://github.com/MarginallyClever/Makelangelo-software/issues/671.

i-make-robots, 2023-01-30:
AFAIK the warnings appear on all platform versions so they're probably not relevant.  
I wonder... does the same problem happen with 7.34?
Unfortunately I no longer have a mac for local testing.

jasonluther, 2023-01-30:
Yes, the error happens with 7.34. 

From Makelangelo-nightly-20230127-7fdf19.zip:
```sh
 % java -jar Makelangelo-7.34.1-with-dependencies.jar
Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: /private/var/folders/_s/rwt5whxs2f11s2qzwz5pd97r0000gn/T/jogamp_0000/file_cache/jln876458083632194203/jln8065683382353028678/natives/macosx-universal/libgluegen_rt.dylib: dlopen(/private/var/folders/_s/rwt5whxs2f11s2qzwz5pd97r0000gn/T/jogamp_0000/file_cache/jln876458083632194203/jln8065683382353028678/natives/macosx-universal/libgluegen_rt.dylib, 0x0001): tried: '/private/var/folders/_s/rwt5whxs2f11s2qzwz5pd97r0000gn/T/jogamp_0000/file_cache/jln876458083632194203/jln8065683382353028678/natives/macosx-universal/libgluegen_rt.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/private/var/folders/_s/rwt5whxs2f11s2qzwz5pd97r0000gn/T/jogamp_0000/file_cache/jln876458083632194203/jln8065683382353028678/natives/macosx-universal/libgluegen_rt.dylib' (no such file), '/private/var/folders/_s/rwt5whxs2f11s2qzwz5pd97r0000gn/T/jogamp_0000/file_cache/jln876458083632194203/jln8065683382353028678/natives/macosx-universal/libgluegen_rt.dylib' (mach-o
``` 

I also tried sudo xattr -cr /Applications/Makelangelo.app
on the .app in https://github.com/MarginallyClever/Makelangelo-software/releases/download/Nightly/Makelangelo-7.34.1-nightly-20230127-7fdf19.dmg, and Mac OS still warns that it's damaged and won't open it. 

It looks like there's some issue with how JOGL is packaged. I spent a few hours this weekend trying to sort it out, but it's way beyond what I know about java/maven/etc. https://github.com/jzy3d/jogl-maven-deployer#getting-jogl-24-rc-4-for-macos-bigsur-20211116 (via https://github.com/jzy3d/jogl/issues/30) seem to be relevant, but I could be wrong. 

I'm happy to try anything I can to assist.

i-make-robots, 2023-01-30:
I imagine it's something to do with the fat jar... which is probably lagging behind a few versions....
Is there any previous executable that works?  JOGL hasn't changed in a long time.  Maybe start with the earliest version and then move forward half - binary search, so to speak.

jasonluther, 2023-02-01:
`Makelangelo-7.31.0-with-dependencies.jar` has the same error. I checked out an older version with `git checkout tags/7.3.4`, and `java -jar target/Makelangelo-7.10.0-with-dependencies.jar` produces a similar error: 
```
Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: Can't load library: /var/folders/_s/rwt5whxs2f11s2qzwz5pd97r0000gn/T/jogamp_0000/file_cache/jln374138118390636241/jln5027049364187157223/natives/macosx-universal/libgluegen-rt.jnilib
```

Perhaps it worked previously on Apple silicon machines through Rosetta, but I haven't been able to figure out how to force `java` to run under Rosetta 2.

i-make-robots, 2023-02-06:
Can you confirm if 7.34.2 fixes things for you?  Should appear any moment now in the Nightly builds.  (commit [af7ed49](https://github.com/MarginallyClever/Makelangelo-software/commit/af7ed4990d9c23374f19eebe7126f591c7d1f2ca))

i-make-robots, 2023-02-06:
Please try https://github.com/MarginallyClever/Makelangelo-software/releases/download/Nightly/Makelangelo-7.34.2-nightly-20230206-ba532b.dmg
and report back.

jasonluther, 2023-02-06:
Running the jar built on master at ba532b5ab5130804f044545d1ff3310573fc9435 works! So I consider this issue resolved. Thanks! 

However, I still can't open the application from that .dmg. I get this error: _“Makelangelo” is damaged and can’t be opened. You should move it to the Trash._ That is after running `sudo xattr -cr /Applications/Makelangelo.app` and opening it by right-clicking and selecting "Open" from the menu.

closed 2023-02-06 by jasonluther

i-make-robots, 2023-02-07:
does this help?  https://mcr.dozuki.com/Guide/1.+How+to+use+Makelangelo+Software+7.32+/39?lang=en#s416