MarginallyClever/Makelangelo-software #714: instant quad tree does nothing.

i-make-robots, 2023-10-19:
Describe the bug
run quad tree instant. no image is produced.

v7.49.0

i-make-robots, 2023-10-19:
working in #713 It sort of works but only when I max out some of the settings.

https://user-images.githubusercontent.com/1464454/276738956-01b08789-6031-4e14-a1f9-9c6f8a90b48f.png

itsMohammedThaier, 2023-10-19:
The "Base cut off" is an absolute factor rather than relative factor

It determines whether a subfractal deserves to be divided further more based on the average darkness of its area

if you try and import an image where the average darkness is low, the algorithm will determine the first fractal to not deserve to be divided(Notice when you increase the 'Base cuf-off' the darkness range increase therefore it allows for more fractal iterations)

Try importing an image that's dark enough and it should look fine

**In short, the image is too 'light*to be captured by the first iteration of the fractal**

i-make-robots, 2023-10-20:
Fixed in #713

closed 2023-10-20 by i-make-robots