MarginallyClever/Makelangelo-software #741: Ability to modify start and end g-code

anghelos, 2024-03-05:
**Describe the solution you'd like**
It would be very useful to be able to modify the start and end g-code that Makelangelo software generates.

While we can add user g-code to the plotter settings, this is just added to the existing code; we can't modify what Makelangelo adds after.

For example, I'd like my plotter to first do a G28, then go to a specific point before asking to insert the pen. With the current state of things, even if I add all that in the start g-code, it will do a G28 all over again right after.

i-make-robots, 2024-03-06:
Saving a bunch of extra gcode lines to app Preferences isn't hard.  Putting them in the right spot for every user sounds like a challenge.

How do you envision the interface?  Please describe how you get to the editor that lets you make this change.  Why is it better to do it in the app than, say, a text editor after exporting to gcode?  

Do you know of an app that already handles this well?  Got a screenshot or video example?  We can copy what they do.

anghelos, 2024-03-06:
> How do you envision the interface? Please describe how you get to the editor that lets you make this change.

I think the easiest would be to keep the existing custom G-code interface, but to have it pre-filled with the default g-code (and have this be the _only_ start and end g-code).

> Why is it better to do it in the app than, say, a text editor after exporting to gcode?

Modifying the g-code in a text editor isn't very user-friendly. I manage a school's makerspace, and would like the experience to be plug-and-play for students, with the polargraph already pre-configured for them.

> Do you know of an app that already handles this well? Got a screenshot or video example? We can copy what they do.

PrusaSlicer uses a similar approach to the one I mentioned above. The interesting thing they do is that they have some predefined variables, and allow some if statements, among other things, but that's outside of the scope of this issue. Here's a screenshot with a non-modified printer: 
[image]

i-make-robots, 2024-03-06:
[src/main/java/com/marginallyclever/makelangelo/makeart/io/SaveGCode.java#208](https://github.com/MarginallyClever/Makelangelo-software/blob/40e50e7f61d269a6fc1630a5997fdaaa9f447624/src/main/java/com/marginallyclever/makelangelo/makeart/io/SaveGCode.java#L206)

The only line that could *easily* be added to custom gcode is the home command.  The color changes are considered steps in the path walked by the turtle.  

https://github.com/MarginallyClever/Makelangelo-software/blob/40e50e7f61d269a6fc1630a5997fdaaa9f447624/src/main/java/com/marginallyclever/makelangelo/makeart/io/SaveGCode.java#L235-L236

Maybe a new custom could be added between pen up and change color?

anghelos, 2024-03-06:
>  The color changes are considered steps in the path walked by the turtle.

Yes, I realized that after I replied yesterday. I also though that going to 0.0 was part of the start sequence, but I'm realizing now that it was my drawing that started from the center.

Honestly, the G28 after the custom G-code is the main thing that I'd like to change. And I'm realizing now that I can probably get around that by using XY_AFTER_HOMING in the Marlin firmware instead. Soooo... very low priority on this one? :)

i-make-robots, 2024-03-06:
Well... I'm AFK from march 7-18.  If someone does it before me?  Cool.

i-make-robots, 2024-03-06:
Also thank you for submitting such a great feature request with details and everything.

i-make-robots, 2024-10-11:
**7.57.1** committed just now lets you adjust the home gcode.  leave it blank to do nothing.  
`Settings > Plotter settings > Custom GCode > Home Gcode` is where you'll find the value.

closed 2025-01-04 by i-make-robots