MarginallyClever/Makelangelo-software #743: Adding multiple objects, with color changes.

anghelos, 2024-03-21:
I'd be great to be able to add multiple objects to the same canvas.

For example, I'd like to be able to add an image, select the style I want for it, then also add text.

These should be able to be moved (translated) independently.
Ideally, they could also be assigned a color, for a tool change prompt.

i-make-robots, 2024-03-21:
A workaround is to generate them separately and then use the "import" option to merge them together.

i-make-robots, 2024-03-21:
The other way is to build a lot of features that exist better already in other art programs link inkscape.

closed 2024-03-21 by i-make-robots

anghelos, 2024-03-21:
> A workaround is to generate them separately and then use the "import" option to merge them together.

Thanks, didn't realize that "import" would not overwrite the existing artwork.

i-make-robots, 2024-03-21:
Think of Load as `new + import`.