MarginallyClever/Makelangelo-software #783: Slider node

laurentbbb, 2025-02-07:
A slider as node could be an interesting way to adjust settings when number are needed, instead of everytime right-click edit
node.

In parameters of this node, it should be possible to set the range (between 0 and 100 for ex) and incrementation (step of 1, 10, 100...) 

[image]

(inspiration from node of Grasshopper in Rhino)

i-make-robots, 2025-02-08:
Part of the challenge is knowing when a user is dragging a whole node and when they are dragging the slider handle...

laurentbbb, 2025-02-08:
I imagine that each time a numeric value is possible to attach such as height, length, space, quantity, thickness... to save time and avoid going into the node edition to change the values

i-make-robots, 2025-02-08:
I believe #777 will help with this.  You'll select a node and the panel will be there, immediately in a Dock, so you don't have to `right click > Edit`.  This way I don't have to invent my own slider code.

i-make-robots, 2025-02-19:
Best i can do at the moment is a slider where there should be a slider, such as:

[image]