MarginallyClever/Makelangelo-software #785: Canvas position

laurentbbb, 2025-02-07:
After creating a drawing with nodes, when creating a canvas, it appears above the drawing, hiding it 

Canvas is supposed to be always under a drawing 

7.65.0
Windows 10

[image]

laurentbbb, 2025-02-07:
A temporary solution is to select all nodes except canvas, cut, paste and force reload

the canvas is then under the drawing

i-make-robots, 2025-02-07:
The `PrintWithGraphics` items will gain a `getLayer()` which controls the order in which things are drawn.  Lower numbers go first, on the bottom.  All classes implementing `PrintWithGraphics` should let the user choose, and have a sensible default value.

i-make-robots, 2025-02-08:
Fixed in `7.66.0`

closed 2025-02-08 by i-make-robots