MarginallyClever/Makelangelo-software #790: Canvas node position

laurentbbb, 2025-02-15:
When creating a Canvas node, a XY value of -500 is directly applied, drawing will be center to left bottom corner missing 3/4 of it.
The values ​​are grayed out and cannot be changed.
Often for other nodes, if the values ​​are not chosen from the start, it is no longer possible to change them after.

7.66.2 Windows 10

[image]

i-make-robots, 2025-02-15:
This is as intended.  
- The paper is centered on the origin, just like it would be on the Preview tab.  
- Outputs should be read-only.
- Inputs that are connected to something upstream should be read-only.

laurentbbb, 2025-02-16:
Example of a drawing not center

[image]

i-make-robots, 2025-02-17:
Please list the steps and provide the source image (drop it here) so I can match your results.

i-make-robots, 2025-02-17:
I have successfully reproduced your results.

i-make-robots, 2025-02-17:
DetectEdges was not generating a result in the right place.
A temporary fix is to add a TransformTurtle to move the result before printing.
Here is an example of the fixed result:
[image]

commit d68b46f0 (2025-02-17): fix #790 and vastly increase PrintTurtle performance
closed 2025-02-17 by i-make-robots