MarginallyClever/Makelangelo-software #801: LCD does not power on

kitplum, 2025-02-21:
I connect the power transformer to the power receptacle on the left of the HUGE.
Press button on LCD screen and the panel does not switch on.

Please advise what I need to troubleshoot

kitplum, 2025-02-21:
[image]
here are my connections

i-make-robots, 2025-02-22:
Try swapping the two grey ribbon cables.  If there is a blue light but no text, try adjusting the contrast knob, which is on the back of the top-left corner of the screen.  if you look in from above, it's a white knob on a blue body.

closed 2025-02-24 by i-make-robots

kitplum, 2025-02-28:
That did not work...what can I test next??  sorry that took me a couple of days to get to...