MarginallyClever/Makelangelo-software #835: Donatello to Preview windows integration

Jackymancs4, 2025-05-17:
**Is your feature request related to a problem? Please describe.**
I really like the Preview window. It gives a good sense of the final print, and has a practical feeling to it with travel moves option and the step slider.

**Describe the solution you'd like**
As a way to connect the Preview window to the Doneatello window, I would like two specific nodes.

*CanvasFromPreview*
It generates a Canvas like the "Canvas" node, but using the Paper settings (and stays synced with it).

*TurtleToPreview*
Like the "PrintTurtle" but it export the finished line to the Preview, where I can use the Step slider to see the evolution of the print (or just get a better feeling of the movements).

**Describe alternatives you've considered**
I saw the issue #778 but it did not make clear to me the vision for the preview panel.

i-make-robots, 2025-05-18:
At present the two systems are still very separate.  donatello and preview are almost on opposite sides of a wall.

I'd like to eventually have all the top-level menu items load a graph in the donatello window showing how everything connects.  the output would go directly to the preview window.  Instead of special code hidden in the system to generate each style, it would load a pre-made file with the graph and fill in the few values most users want to tweak (like source image).

Jackymancs4, 2025-05-18:
If you mean that all the tasks in "Generate art" will insert a Graph in the Donatello window, leaving the Preview untoched and the user have connect it to a proper output node (like the TurtleToPreview node I proposed), I think it woul be very cool.

A little complicated is for the tools menu, I guess. Click on flip will it add a flip node right before the output node linked to the preview?

Anyway, do you think that in the short/medium time span, the solution I proposed make sense?

i-make-robots, 2025-05-21:
> If you mean that all the tasks in "Generate art" will insert a Graph in the Donatello window, leaving the Preview untoched and the user have connect it to a proper output node (like the TurtleToPreview node I proposed), I think it woul be very cool.

Exactly.  The origin of Donatello was that I built many generators and they duplicate a lot of very similar code.  I wanted a way to remove the duplication and increase creative freedom.

> A little complicated is for the tools menu, I guess. Click on flip will it add a flip node right before the output node linked to the preview?

A TransformTurtle that in normally default (do nothing).

> do you think that in the short/medium time span, the solution I proposed make sense?

Cautiously yes.  I am concerned that it leads to problems like ... if I have a donatello program with a `TurtleToPreview` and then I use File > Load it will convert an image and display it in the preview without touching Donatello.  It will last until the the `TurtleToPreview` is refreshed and then disappear, confusing the user.

I think the better solution is to convert the generators, and make the `File > Load` into 

1. ...a process that loads a Dontallo graph OR 
2. ..opens a wizard that walks through basic Donatello creation (probably very hard to do well, closer to the existing system). 

And in both cases all documentation, tutorials, and translations would need to be updated.  Fun!