MarginallyClever/Makelangelo-software #836: Key bindings in Donatello window do not work

Jackymancs4, 2025-05-17:
**Steps to make the bug happen**
1. Go to Donatello windows
2. Add node, and select it
3. Press Ctrl+E 
4. See the estimate time, instead 

**What was supposed to happen**
The shortcuts suggested in the contextual menu should do what they advertise.

**Platform (please complete the following information):**
 - Version: master
 - OS: Windows

**Additional context**
Also Ctrl-Z does not work. I tried to delete a node, while all where selected (without me realising it) and accidentaly trashed everything. I was unable to undo it.

i-make-robots, 2025-05-18:
Either 
a) not show shortcuts on the actions in the popup menu or 
b) make a top-level menu that includes the same commands so that they work as intentded.

B is probably the better way to go.

Jackymancs4, 2025-05-18:
The accelerators really are useful. I guess the in case of conflicts (like Ctrl+E) if the Donatello window has the focus it should prevail. Use tha more "global" shortcut otherwise.

For the Ctrl-Z problem ("Additional context" section) would you like a separate issue?

i-make-robots, 2025-05-20:
In the case of conflicts it should be fixed to not conflict, because the top level menu controls everything all the time.

A nice to have would be a place where a user could configure their own accelerators.

CTRL-Z conflicts with another command?