MarginallyClever/Makelangelo-software #838: Customizable Action Accelerators

i-make-robots, 2025-05-21:
**Is your feature request related to a problem? Please describe.**
Sometimes two Actions can have conflicting (identical) accelerators (shortcut keys).
It's impossible to have context-aware accelerators.

**Describe the solution you'd like**
A configuration panel that would list all actions and their accelerators with controls to reset and edit.  In the case of conflicts both items should indicate it somehow - red text or background color, perhaps.

**Additional context**
This would probably be done with a new ActionFactory in Donatello.  All Actions would be created through the factory.  Then the configuration panel could query the entire list etc.

Right-click context-sensitive menus should exist somewhere in the main menu.  Right clicking would simply open the appropriate submenu at the cursor position.

i-make-robots, 2025-05-21:
@Jackymancs4 does that seem sane to you?

i-make-robots, 2025-05-21:
Something like this....
[image]

i-make-robots, 2025-05-21:
when two valid fields match 
[image]

i-make-robots, 2025-05-21:
When an invalid field is used it turns red.
[image]

i-make-robots, 2025-05-21:
https://github.com/MarginallyClever/Donatello/compare/Action-Configuration?expand=1d

Jackymancs4, 2025-05-21:
Looks great. I'll test it tomorrow and I'll edit this comment.